riscv/riscv-debug-spec #854: Need calrification on how anynonexistent bit works

abhi-valtrix, 2023-07-07:
According to the spec - if I select a hart that doesn't exist ; anynonexistent bit in DMSTATUS should get set to 1. I am using spike and open ocd but this doesn't seem to be working as expected

Open On-Chip Debugger
> riscv dmi_write  0x10 0x60001               - trying to select hart 6 in a 4 hart machine.
> riscv dmi_read  0x11          
0x400382                                                   - anynonexistent is not set

Am I understanding the spec wrong ? Please advise

Thanks,
Abhiram

bdwyatt, 2023-07-07:
In a 4 hart system you may only be able to write 2 bits of hart_sel (depends on the specific implementation), which may prevent you from selecting a nonexistent hart. 

You should be able to read back the dmcontrol register and see if the hart_sel value you attempted to write matches the current value.

abhi-valtrix, 2023-07-07:
Thank you @timsifive  and @bdwyatt for your help! however even with poll off i am facing the same issue.

> poll
background polling: on
TAP: riscv.cpu (enabled)

> poll off
> riscv dmi_write 0x10 0x60001 
> riscv dmi_read  0x11         
0x400382

However I find that writing anything greater than the number of cpus results in mhartid going to the highest index for instance in the above case mhartid becomes 3 ( my system has 4 cpus )

> riscv dmi_write 0x17 0x320f14
> riscv dmi_read  0x4          
0x3

timsifive, 2023-07-07:
Oh, I missed this part in the code: https://github.com/riscv-software-src/riscv-isa-sim/blob/master/riscv/debug_module.cc#L809

So that all looks right.

abhi-valtrix, 2023-07-07:
@timsifive  - Okay , so it is always a min(written val , actual no of processors ) - and so hartsel is never written with a value more than the actual number of processors - hence anynonexistent is never set. That makes sense ( as to why it is not working as expected )  :) ... thanks Tim !

abhi-valtrix, 2023-07-07:
Closing this as resolved.

closed 2023-07-07 by abhi-valtrix