riscv/riscv-debug-spec #857: (External) Debugger Implementation Read/Write Memory - transfer not specified in examples

JekkuChen, 2023-07-17:
In the Debugger Implementation, there are step by step instructions regarding how to read/write registers/memory using the Program Buffer. In all the examples, the transfer bit should be set when using the Program Buffer.

In the "Using Program Buffer" section of "Accessing Registers" it shows for example: 
[image]

However, in the Memory Read/Writes, the command register - using Program Buffer - writes seem to be missing specifying transfer to be set which is required

[image]
<sub>Use Program Buffer for Read</sub>  
[image]
<sub>Use Program Buffer for Write</sub>

All memory Prog Buffer examples above don't specify transfer - explicitly -  as set, unlike the register read/write Program Buffer examples which explicitly display that transfer is set in those cases.

timsifive, 2023-09-26:
Thanks for pointing this out.

pdonahue-ventana, 2023-10-06:
This was addressed in #886.

closed 2023-10-06 by pdonahue-ventana