riscv/riscv-debug-spec #863: Program buffer & requirement to insert only single instruction - how to interpret?

JanMatCodasip, 2023-08-15:
How shall we interpret the highlighted text from the spec, telling that the debugger is required to insert no more than one instruction into the program buffer? Is it:

a) a general requirement, applicable to any progbufsize?
b) a requirement applicable only to the case of progbufsize == 1?

My understanding is that b) is right but I'd like to get a confirmation.

Depending on the outcome of this discussion, I'd submit a change to clarify that text.

Thank you.

[image]

pdonahue-ventana, 2023-08-15:
I think that you're right.  It basically means:
> If progbufsize is 1 then:
> - impebreak must be 1
> - It is possible that the Program Buffer can hold only one 32-or 16-bit instruction, so the debugger must only write a single instruction in this case, regardless of its size. This instruction can be a 32-bit instruction, or a compressed instruction in the lower 16 bits accompanied by a compressed nop in the upper 16 bits.

If the last two sentences were independent of progbufsize then there would be no point in allowing progbufsize>1.

JanMatCodasip, 2023-08-16:
@pdonahue-ventana, thank you for your response.

I have submitted a pull request that should make the text clearer: https://github.com/riscv/riscv-debug-spec/pull/868

JanMatCodasip, 2023-08-25:
Closing this issue -- it is resolved by https://github.com/riscv/riscv-debug-spec/pull/868.

closed 2023-08-25 by JanMatCodasip