riscv/riscv-debug-spec #873: Convert LaTeX specification to AsciiDoc.

wmat, 2023-09-12:
Opening this issue to track the conversion of the LaTeX specification to AsciiDoc.  Conversion will be done on the [asciidoc branch](https://github.com/riscv/riscv-debug-spec/tree/asciidoc).

timsifive, 2023-09-13:
Hi Bill,

The trickiest part is probably dealing with the way we handle registers. Their definitions are in https://github.com/riscv/riscv-debug-spec/tree/master/xml and there is some LaTeX markup mixed in there as well. Then https://github.com/riscv/riscv-debug-spec/blob/master/registers.py takes that XML and turns it into LaTeX figures/tables. Modifying that script is probably not too bad, to change the output to asciidoc + wavedrom.

The way this project has been presented to me is that you'll do the straightforward 80% of this project, leaving the remaining 80% to debug group. Is that right?

Tim

wmat, 2023-09-14:
That's correct, I'll do the bulk of the conversion but hope that things
like tweaking the scripts to output asciidoc instead of latex can be done
by the debug spec authors.

timsifive, 2023-09-14:
If you can convert some of the LaTeX output for one file into the relevant asciidoc and wavedrom so I have something to go on, I can change the script to output that format.

E.g. `make dm_registers.tex` and then show me what that file should look like up to (not including) the dmcontrol subsection.

wmat, 2023-09-25:
Hi Tim,

I'm just getting back to your spec now.  In looking at the src tree, I
don't see a dm_registers.tex file.  I see it referenced in the Makefile but
don't see it in the src.  Am I looking in the wrong place?

Thanks
Bill

timsifive, 2023-09-25:
`make dm_registers.tex` should get you the file. I'll attach it here as well, renamed to .txt otherwise github won't let me.
[dm_registers.txt](https://github.com/riscv/riscv-debug-spec/files/12717362/dm_registers.txt)

wmat, 2023-09-25:
FWIW, when I do make dm_registers.tex the result is an empty file even
though it appears to have worked.  I can just use your file you attached.

make dm_registers.tex
make: 'dm_registers.tex' is up to date.

timsifive, 2023-09-25:
The Makefile isn't very good. Probably the file was generated but there was an error while generating it. So now `make` won't regenerate it because it already exists.

wmat, 2024-02-16:
@rtwfroody just want to ask if there's anything left to do for the asciidoc conversion?

rtwfroody, 2024-02-16:
I think it looks good, although I look forward to bringing in a list of table and figures once you figure out the best way to do that.

I might tweak the diagrams more at some point (really wish I could left/right align the bit offsets), but they're good enough.

wmat, 2024-02-16:
Whenever you open an issue around formatting, feel free to @ me so I don't miss it.  If I can help, I certainly will.