riscv/riscv-debug-spec #877: Is an address mcontrol/mcontrol6 trigger valid to compare subset of the other addesses?

YenHaoChen, 2023-09-18:
Hello, 

The specification requires an address mcontrol/mcontrol6 trigger to compare the lowest address and recommends comparing other accessed addresses. For example, consider a 4-byte memory access with the lowest address 0x4000. An address trigger is required to compare address={0x4000} and recommanended to compared address={0x4000, 0x4001, 0x4002, 0x4003}.

Is it valid to compare an arbitrary subset of the other addresses? For the above case, is comparing address={0x4000, 0x4002} only valid? (Although this does not make sense from a software perspective, I cannot find a solid answer from the text of the specification.)

[image]

Reference: https://github.com/riscv-software-src/riscv-isa-sim/pull/1459

timsifive, 2023-09-19:
My reading is that they are all valid. However, if you want to make a product that is useful to people then design C would be very confusing. The spec is intentionally a bit vague here because implementing triggers can be expensive, because comparators are sizable and might be on the critical path. It's better to implement something suboptimal than to implement nothing at all.

closed 2023-09-19 by YenHaoChen

pranjal20498, 2024-05-28:
if my tdata2 is the same as any load address value then it would take trigger for match=0?

rtwfroody, 2024-05-30:
@pranjal20498, yes, that is the preferred behavior.