riscv/riscv-debug-spec #889: Triggers for interrupts are unclear when AIA is added

jhauser-us, 2023-09-30:
Section 5.6.14, "Interrupt Trigger", is unclear about how to respond to interrupts that may be injected into S or VS mode using new features added by extension Smaia/Ssaia, the RISC-V Advanced Interrupt Architecture (AIA).

Let's first ignore the AIA and just look at the hypervisor (H) extension, which predates the AIA.  The H extension defines interrupt numbers 2, 6, and 10 as VSSI, VSTI, and VSEI at M and S/HS levels, but when delegated down to VS mode, these get translated to interrupt numbers 1, 5, and 9 respectively, which are SSI, STI, and SEI.  If I want to have a trigger on an STI (interrupt number 5) at VS level, am I right to assume I must set bit 6 (VSTI) in `tdata2`?  Or should I set bit 5 (STI) in `tdata2`?  I think the answer is intended to be bit 6 (VSTI), but some clarification is in order.

The AIA upends the Debug spec's interrupt triggers by adding the ability to inject interrupts into lesser privilege modes that completely bypass `mip` and `mie`.  For example, using register `hvictl`, a hypervisor can inject into VS mode an interrupt number 5 that has no connection to bits 5 or 6 in `mip`/`mie`.  If I want to have a trigger for interrupt number 5 at VS level caused by `hvictl`, should I again set bit 6 (VSTI) in `tdata2`, or should I set bit 5 (STI)?

If consistency is maintained with past practice and bit 6 of `tdata2` always indicates interrupt number 5 at VS level, then what about the case when a hypervisor injects into VS mode interrupt number 6?  (This is something a hypervisor may reasonably do, if, for example, it's emulating the H extension for a nested hypervisor running as a guest in VS mode.)  To configure a trigger for interrupt number 6 at VS level, what bit should I set for that in `tdata2`, if it's been decreed that bit 6 in `tdata2`means interrupt number 5 at VS level?

On a related note, if AIA registers `mvien` and `mvip` are used to inject an arbitrary interrupt number X into S level (no involvement of the H extension here), then whether interrupt X is enabled at S level has nothing to do with bit X in `mie`.  But Section 5.6.14 says:
> The trigger only fires if the hart takes a trap because of the interrupt. (E.g. it does not fire when a timer interrupt occurs but that interrupt is not enabled in `mie`.)

Using the AIA's `mvien` and `mvip`, a timer interrupt (or any other interrupt) may be enabled for S mode---and actually taken into S mode---even though the bit for that interrupt number is zero in `mie`.  The same of course applies for interrupts that a hypervisor may inject into VS level using AIA registers `hvien` and `hvip`, or the aforementioned `hvictl`.

It seems to me, a complete resolution for the cases where an interrupt number changes when delegated down to VS level will be difficult to devise unless an `itrigger` configuration can indicate the privilege level to interpret the bits of `tdata2`.  (It's not fully unambiguous simply to associate the bits of `tdata2` with the interrupt number to be written to the `*cause` CSR when a trap is taken, because this value can change depending on which privilege mode ends up taking the trap.)  But currently, `itrigger` specifies nothing for the privilege mode into which a trap is taken, only for the privilege mode *from which* a trap occurs.  Unfortunately, this isn't sufficient information when combined with the facilities added by the AIA for raising interrupts that bypass `mip`/`mie`.

timsifive, 2023-10-05:
I'm not clear on exactly what you mean by an interrupt "at" a level. Interrupts are taken *from* one level *into* another (which may be the same).

Regardless, the rule should be: The bits written in tdata2 correspond to the number the interrupt has in the mode that it is taken *from*. Does that sound reasonable?

Is it still true that every interrupt taken from a given privilege mode has a number that used for that interrupt in that privilege mode? If not, I'm inclined to say that `itrigger` currently does not fire on those interrupts. We can extend the trigger in the future. I would rather not do that now, because this whole process has gone on for long enough.

jhauser-us, 2023-10-05:
> I'm not clear on exactly what you mean by an interrupt "at" a level.

For each privilege level that can take an interrupt trap (M, S/HS, and VS), interrupts are visible "at" that level through the `*ip` bits of that level, namely `mip` for M level, `sip` and `hip` for S/HS level, and `vsip` for VS level.  A hypervisor, for instance, can use its CSRs `hvien`, `hvip`, and/or `hvictl` to make interrupts appear in `vsip` at VS level.

> Regardless, the rule should be: The bits written in tdata2 correspond to the number the interrupt has in the mode that it is taken from. Does that sound reasonable?

I don't know that that's well-defined.  Assume we're executing in VS mode and an SEI (interrupt number 9) occurs, causing a trap from VS mode to M mode or HS mode.  What do you say is the number that this interrupt has in VS mode?  9, I presume?

Now, what if we're executing in VS mode and a VSEI (number 10) occurs to cause a trap that gets delegated to VS mode.  At VS level, this interrupt gets translated into an SEI.  What do you say is the number that this interrupt has in VS mode?  9 again, right?  It can't be 10.

If that's correct, then evidently bit 9 in `tdata2` will refer to both SEIs and VSEIs when executing in VS mode.  So, will bit 10 in `tdata2` have any effect when executing in VS mode?

> Is it still true that every interrupt taken from a given privilege mode has a number that used for that interrupt in that privilege mode?

I would say "no".  If an MTI causes a trap from S mode or VS mode to M mode, the MTI is interrupt number 7 at M level, but interrupt number 7 is officially reserved at S and VS levels.  S and VS levels don't even have a concept of machine-level traps, so how can we say they assign the same numbers to these traps as M level does?

I believe that applying that logic to your proposed rule ("`itrigger` currently does not fire on those interrupts") would mean that M-level interrupts do not match any interrupt trigger when a hart is executing in modes less privileged than M.  As I read the Debug Spec, that would be a non-compatible change in the behavior of interrupt triggers.

If the TG is sure it knows how it wants to answer these questions, then maybe there's a way to specify its desired solution.  I need to hear the answers before I can attempt to judge.

timsifive, 2023-10-05:
> If that's correct, then evidently bit 9 in tdata2 will refer to both SEIs and VSEIs when executing in VS mode. So, will bit 10 in tdata2 have any effect when executing in VS mode?

I'm totally fine with that.

> I believe that applying that logic to your proposed rule ("itrigger currently does not fire on those interrupts") would mean that M-level interrupts do not match any interrupt trigger when a hart is [...]

That's a good point. I'll have to ponder this a bit more.

Choosing the interrupt number in the destination mode might solve this, but it's more confusing because the mode in itrigger is defined to be the from mode. I also imagine that now the debugger might have to choose different interrupt numbers based on which more privileged modes are implemented, which is also awkward.

pdonahue-ventana, 2023-10-05:
These are all valid points.  The itrigger mechanism existed in the ratified 0.13 spec but it was obviously not updated to deal with more recent extensions (except for the trivial change of adding vs and vu bits).

It's been several years so I just refamiliarized myself with what ARM does.  In short, they don't have itrigger.  They simply have a way to halt on all synchronous exceptions to a particular mode (Exception Catch).  That's equivalent to having no itrigger at all and having etrigger that has tdata2 hardwired to all ones.  They do have a deprecated feature (Vector Catch) that's like itrigger but it's only supported in AArch32 for self-hosted debug (for backward compatibility with old 32-bit debuggers).  So I suppose that they looked at how it was being used and decided that itrigger isn't useful enough to carry forward.

I'm not arguing for itrigger to go away.  I just intended to see how others solved this problem.  ARM solved the problem by never allowing it to exist to start with.

timsifive, 2023-10-09:
Use cases for itrigger:
1. First instruction of the trap handler can't be fetched for some reason. This lets a user debug that.
2. Want to set a hardware breakpoint at the start of a trap handler. itrigger is cheaper to implement than mcontrol6.

So we are mostly concerned with the trap handler, not with the code that was running when the interrupt occurred. But the current mode bits in itrigger designate *from*. That must remain the same.

I think that means itrigger fires when:
1. A trap is taken due to an interrupt from a privilege mode whose mode bit is set in itrigger.
2. AND the tdata2 bit corresponding to the value written to the Exception Code field of the mode-relevant `cause` register is set.

How does that sound?

pdonahue-ventana, 2023-10-11:
> I think that means itrigger fires when:
> 
> 1. A trap is taken due to an interrupt from a privilege mode whose mode bit is set in itrigger.
> 2. AND the tdata2 bit corresponding to the value written to the Exception Code field of the mode-relevant cause register is set.
>
> How does that sound?

That makes sense to me.  I believe that this allows you to handle any of your use cases.  For example, if you're debugging the VS mode's timer interrupt handler then you set tdata2[5] and the VU/VS bits.  Then it doesn't matter whether this interrupt comes via a virtual interrupt from a higher level, whether this interrupt is considered to be 6 instead of 5 at a higher level, etc.  It will enable you to debug the interrupt that this VS handler considers to be interrupt 5, that vstvec.mode=vectored would vector to vstvec+4*5, etc.

Because the mode bits are the from mode instead of the to mode, I think that you sometimes get unfortunate things.  One example is when debugging the S/HS STI handler.  You'd have to set tdata2[5] and the S/U/VS/VU bits since I think that you can come from any of those modes to the S/HS STI handler.  Note that this will also catch interrupts going to VS mode (the example above).  Presumably, a debugger can see that this has happened (mstatus.MPP or dcsr.prv is VS) and immediately resume if it's not the handler you're interested in.  This isn't ideal but it's better to have some false positives than to miss the interrupts you care about.  I don't think that there's a better way to do this without breaking backward compatibility.