riscv/riscv-debug-spec #891: What purpose does the external trigger input named intctl serve?

jhauser-us, 2023-10-01:
As described in Section 5.6.16, an external trigger can match on any of 16 implementation-specific external signals, and/or on "intctl", which is described as a signal from an attached interrupt controller.  I can find nothing to illuminate what purpose "intctl" serves that is any different from the other 16 implementation-specific external trigger inputs.

The document should explain why the signal labelled intctl needs to be standardized as distinct from the 16 other external trigger inputs.  Exactly under what standard-specified circumstances may software expect intctl to signal?  The existing documentation for standard RISC-V interrupt controllers, PLIC, APLIC, and IMSIC, doesn't suggest to me any use for intctl, so what am I missing?

pdonahue-ventana, 2023-10-02:
This was added for CLIC.  See https://github.com/riscv/riscv-debug-spec/issues/412.

(It's also funny to see that we thought that we were freezing this spec 3 years ago under the then-current rules.)

pdonahue-ventana, 2023-10-06:
#898 incorporates the answer into the spec.

closed 2023-10-06 by pdonahue-ventana