riscv/riscv-debug-spec #893: Why does an instruction count trigger also count traps?

jhauser-us, 2023-10-01:
Section 5.6.13, "Instruction Count", says that an instruction count trigger decrements its count on any trap, even if no instruction retires.  This is odd to me, given that some care is taken otherwise to decrement the count only for instructions that retire after being fetched in a matching privilege mode, and no other instructions.  At a minimum, I think a comment is warranted to explain the choice to decrement the count on traps, especially for those traps that the Unprivileged Spec categorizes as *invisible* (from Section 1.6, "Exceptions, Traps, and Interrupts").  Decrementing the count on invisible traps seems to me to flout the intention behind the word *invisible*.

timsifive, 2023-10-02:
The core reason to count all traps is that a debugger wants to retain control when an instruction is executed which causes a trap that should be resolved but isn't. E.g. a page fault, and the fault handler restarts the instruction without configuring the proper page table entry. Ignoring invisible traps sounds appealing, but I worry that it makes debugging harder exactly when it's most needed.

I didn't realize invisible traps were a thing. I think what matters in this case is whether the traps are visible to the debugger which set the trigger. Since triggers are only settable from Debug Mode or M-Mode, all traps presumably are visible. Is that right?

pdonahue-ventana, 2023-10-06:
#896 incorporates the answer into the spec.

closed 2023-10-06 by pdonahue-ventana