riscv/riscv-debug-spec #904: For MXLEN=32, can't trigger on interrupts 32 or higher

jhauser-us, 2023-10-18:
The RISC-V Advanced Interrupt Architecture provides for the possibility of "major" interrupts up to number 63, even when XLEN = 32.  The AIA itself allocates interrupt numbers 35 and 43 for low-priority and high-priority RAS events, and other RISC-V extensions may define other interrupt numbers in the range of 32 to 47.

When MXLEN/DXLEN = 32, it appears that an interrupt trigger cannot select interrupt numbers 32 and higher to match on, because `tdata2` is only 32 bits wide.  This limitation probably deserves a comment at a minimum.

pdonahue-ventana, 2023-10-18:
This seems a bit more troublesome than the etrigger problem because there are standard interrupts in that high half.  Part of me wonders how many RV32-only implementations will have RAS.  The problem still exists but maybe it's not a critical flaw in practice.

I think that there's space for a tdata4 at 0x7a6 or 0x7a7 but my inclination is to not introduce that right now.  I think that we should have a comment so that implementations can keep this in mind and any future revisions of the debug spec could address the problem.

timsifive, 2023-10-18:
I agree. Comment and leave it for the future to properly address (if we decide it's necessary).

commit 55ab2a96 (2023-10-18): AR: Comment itrigger limited to 32 ints if XLEN=32

Resolves #904.
commit cc901cfb (2023-10-19): AR: Comment itrigger limited to 32 ints if XLEN=32

Resolves #904.
PR #906 (2023-10-19): AR: Comment itrigger limited to 32 ints if XLEN=32
Resolves #904.