riscv/riscv-debug-spec #905: For MXLEN=32, can't trigger on exceptions 32 or higher

jhauser-us, 2023-10-18:
The RISC-V Privileged ISA 1.13 will expand `medeleg` to 64 bits, in anticipation of future standard exception cause codes in the range 32 to 47.  When MXLEN = 32, a new high-half CSR, `medelegh`, will provide access to bits 63:32.

When MXLEN/DXLEN = 32, it appears that an exception trigger cannot select exception numbers 32 and higher to match on, because `tdata2` is only 32 bits wide.  This limitation probably deserves a comment at a minimum.

pdonahue-ventana, 2023-10-18:
My understanding is that medelegh is providing room for future expansion but that priv 1.13 doesn't introduce any standard exception cause values in that high half.  However, this could be a problem for implementations that have custom values in that high half (48-63).

My inclination is to add a note that we have this limitation and to point out that implementers should try to work within the 8 custom exception causes in the 24-31 range if possible.  The only other thing that immediately comes to mind is to add a tdata4 CSR but I don't think that we want to do that for this case.  While it is undoubtedly better to have standard ways to deal with custom exceptions, implementations with custom exceptions could theoretically also have a custom way to deal with this.

timsifive, 2023-10-18:
I agree. Comment and leave it for the future to properly address (if we decide it's necessary).

timsifive, 2023-10-18:
> The RISC-V Privileged ISA 1.13 will expand `medeleg` to 64 bits, in anticipation of future standard exception cause codes in the range 32 to 47. When MXLEN = 32, a new high-half CSR, `medelegh`, will provide access to bits 63:32.

Is there a PR or an issue or something we can link to as a reference?

pdonahue-ventana, 2023-10-18:
https://github.com/riscv/riscv-isa-manual/pull/1109

timsifive, 2023-10-18:
Thanks. This shows an expansion of `medeleg` but not yet the actual expansion of exception cause codes. I guess that's all there is for now.