riscv/riscv-debug-spec #913: Behaviour of triggers after hart reset

kr-sc, 2023-11-23:
Recently I faced with an OpenOCD issue: wp/bp are not cleared after `reset` command. I tried to investigate is this behaviour correct or not. In my opinion, if hardware triggers are (not) cleared after reset, internal OpenOCD wp/bp should be (not) cleared too. I tried to find some information about it in debug-spec, but I didn't find anything. So, my question is: should hardware triggers be cleared after reset command?

kr-sc, 2023-11-23:
Note: I faced with question when I tried to investigate an OpenOCD issue: wp/bp are not cleared after reset command. I tried to investigate is this behaviour correct or not. In my opinion, if hardware triggers are (not) cleared after reset, internal OpenOCD wp/bp should be (not) cleared too.

pdonahue-ventana, 2023-11-27:
What reset are you talking about?  Did OpenOCD do hartreset?  Did OpenOCD do ndmreset?  Did someone else reset the hart?  I'm trying to understand if the debug spec needs a clarification or if this is an OpenOCD problem that should be tracked at https://github.com/riscv/riscv-openocd.

aap-sc, 2023-11-27:
@pdonahue-ventana this is about what happens with the triggers after `ndmreset` or `hartreset` . We've tried to find to find this information in `Sdtrig` and `Sdext` sectiosn of the latest spec and could not find any relevant information

aap-sc, 2023-11-28:
> That clearly includes the trigger module for each hart. If reset is applied to the TM then you'll return to the reset values specified in Sdext and Sdtrig.

The thing is that Sdtrig does not  seem to **define** the state of TM after reset. Whatever values `tinfo`, `tdataX` after reset are irrelevant since they begin to make sense only after some valid value is written to tselect. So the question boils down to: "what state does TM have after the reset?"

pdonahue-ventana, 2023-11-28:
I thought that the question was about which reset domain the TM is in.  It seems that the question is about what the reset values are.

> The thing is that Sdtrig does not seem to define the state of TM after reset.

I don't understand.  Are you looking at https://github.com/riscv/riscv-debug-spec/blob/master/riscv-debug-stable.pdf?  There are "Reset" columns in the various tables and those reset values come from [the XML](https://github.com/riscv/riscv-debug-spec/blob/master/xml/hwbp_registers.xml).  Can you give a more specific example of something that you don't think is specified?

> Whatever values tinfo, tdataX after reset are irrelevant since they begin to make sense only after some valid value is written to tselect.

It seems relevant.  Even if tselect weren't reset to 0, it would be required to be reset to a legal value (as all WARL registers are).  The first instruction after reset could read tdata1 and it would see `trigger[tselect].tdata1` which for each implementation has a defined reset value for all values of tselect.  The reset value of `trigger[tselect].tdata1[XLEN-1:XLEN-4]` is an implementation-specific legal value and the reset value of `trigger[tselect].tdata1[XLEN-5:0]` is architecturally defined based on the value of those top 4 bits.  Then you can repeatedly write different values to tselect and observe the reset values of the other triggers without having to first write them (which is what it means to have a reset value).

Even if tselect weren't reset to a legal WARL value (and now we're getting even further into the theory and less into what's specified), you could still observe that the underlying triggers were reset by writing to tselect and then reading tdata1 without first writing to tdata1.

> So the question boils down to: "what state does TM have after the reset?"

Based on a quick grep of the XML: The icount.count field is 1, there are a few implementation-defined (Preset) fields, and everything else is 0.

A different debug architecture could have had more minimal reset requirements, more like the privileged architecture does.  Maybe you could get away with only resetting the mode enable bits in all of the triggers to 0, icount.pending to 0, all of the dmode bits to 0, and tmextrigger.select to 0 (and maybe a few more).  But this takes a more maximal reset approach than the privileged architecture and resets almost everything to 0.

aap-sc, 2023-11-28:
> I thought that the question was about which reset domain the TM is in.

No, no! Nothing like that :)

> I don't understand. Are you looking at https://github.com/riscv/riscv-debug-spec/blob/master/riscv-debug-stable.pdf? There are "Reset" columns in the various tables and those reset values come from [...]

Yes, I am looking at this document. The way I read the spec is the following:

* Sdtrig defines a new non-isa extension that defines Trigger Module functionality. Trigger Module can be seen as a collection of triggers that can fire if certain conditions are met.
* Trigger Module has an internal state. This state can be defined as a state of all triggers.
* The spec defines an interface between a hart and TM (trigger module). This interface consists of `tselect` , `tinfo` and `tdataX`.
* My point is that reset values of **registers** `tselect`, `tdataX` does not represent the state of TM after reset.

Why this is important. Consider the following scenario:

- after power on we have an operating hart. Let's assume that TM has no triggers enabled in this state (that is all triggers are disabled after power on)
- then we connect with a debugger,  upload some application on the target and do some debugging
- a debugger sets some hardware triggers to serve as HW breakpoints
- user finishes debug session and resets a hart
- and here comes the question: does the state of TM reset automatically to some platform defined state? Or is it responsibility of a debugger to make sure that it clears triggers it used before the reset was issued?

pdonahue-ventana, 2023-11-28:
> The way I read the spec is the following:
> - Sdtrig defines a new non-isa extension that defines Trigger Module functionality. Trigger Module can be seen as a collection of triggers that can fire if certain conditions are met.

Sdtrig is an ISA extension.  See the name of the chapter and the first sentence.

> - Trigger Module has an internal state. This state can be defined as a state of all triggers.

Yes.

> - The spec defines an interface between a hart and TM (trigger module). This interface consists of tselect , tinfo and tdataX.

The TM is part of the hart.  See figure 2.1.  This might be the source of confusion here.

> - My point is that reset values of registers tselect, tdataX does not represent the state of TM after reset.

When the hart is reset, the hart state must be reset as specified.  For instance, the privileged spec requires that hgatp.MODE is reset to 0 and those bits might live in the memory management unit of your hart.  Similarly, the debug spec requires that various CSRs are reset to 0 even if those bits live in the trigger module unit of your hart.

aap-sc, 2023-11-29:
> Similarly, the debug spec requires that various CSRs are reset to 0 even if those bits live in the trigger module unit of your hart

Ok. But. Trigger module does not have any CSRs. The state of Trigger Module can be **altered** by the hart by accessing some of **harts** CSR. In any case - I don't see where debug spec **defines**/**specifies** what happens with the Trigger Module upon reset.

aap-sc, 2023-11-29:
For example, we have spike as a reference simulator. It just happens that it does not reset trigger state upon **hartreset**. Likely, this is a bug. However I can't find an explicit clarification of that in the debug spec.

aap-sc, 2023-11-29:
> The TM is part of the hart. See figure 2.1. This might be the source of confusion here.

This somewhat helps, but I don't like my conclusion :(.

>  For instance, the privileged spec requires

Unless I'm missing something here is what the spec says:

> Upon reset, a hart’s privilege mode is set to M. The mstatus fields MIE and MPRV are reset to 0. The pc is set to an implementation-defined reset vector. The mcause register is set to a value indicating the cause of the reset. All other hart state is undefined.

ROI is the last sentence.

So if I read this right, then the state of Trigger Module is undefined after reset. I don't think this is expected. Or am I missing something?

pdonahue-ventana, 2023-11-29:
> So if I read this right, then the state of Trigger Module is undefined after reset. I don't think this is expected. Or am I missing something?

The privileged spec only defines the things within the privileged spec.  For instance, Smstateen defines additional CSRs [and says](https://github.com/riscv/riscv-state-enable/blob/main/content.adoc?plain=1#L174) "On reset, all writable mstateen bits are initialized by the hardware to zeros."  The privileged spec's vagueness doesn't override Smstateen's specificity.  Smstateen's specificity overrides the privileged spec's vagueness.  Similarly, the debug spec mandates additional requirements beyond what the privileged spec says.

aap-sc, 2023-11-29:
> Similarly, the debug spec mandates additional requirements beyond what the privileged spec says.

Ok, but could you please point out where does debug spec defines what happens with the Trigger Module state upon reset? We could not find it.

aap-sc, 2023-11-29:
> For example, tcontrol is reset to 0 as below. Section 1.3.3 explains these tables and how to interpret the fourth column.

But these are CSRS of a hart. These do not represent the state of Trigger Module! My understanding is that the values of these CSRs make sense/accurately represent the state of a specific trigger only when a valid index is programmed to tselect.

Do you imply that every trigger should change it state according to these values? This, likely make sense, but I would like to suggest to change the text of the specification to be more explicit.

>  Maybe Spike resets TM state on some resets but not others and maybe hartreset is connected to one of the resets that doesn't reset TM state.

Does the spec allows such behavior?

pdonahue-ventana, 2023-11-29:
What specific state do you consider to be part of the Trigger Module and that you want to know the reset value of?

aap-sc, 2023-11-29:
> What specific state do you consider to be part of the Trigger Module and that you want to know the reset value of?

+CC @timsifive , this may be relevant for OpenOCD

That's a tough question, unfortunately. I'll try to summarize my concerns:

1. The spec defines reset values for some CSR's like `tdata1` as either having constant value or `Preset`. My understanding is the `Preset` can translated as "defined by the platform". We don't have reset value for `tselect`. So this means that the spec implicitly defines **some** (not whole, see below) state of maximum one trigger.

I'd like to suggest the spec to explicitly stat that:

- after reset (`ndmreset`, `hartreset` if you need specifics), each trigger can be in **platform defined** state.
- ideally, I would go as far as to request that each trigger should be **disabled** and not active. However, this may be an unfeasible requirement. Maybe the spec could recommend such an implementation?

2. It looks like currently there is no documented way for a program or an external debugger to figure out if a trigger is active. Currently, the only reliable way (that is the one that always works) to ensure that trigger is disabled is to write "0" in `tdata1`. There is no reliable and portable way to read enable/disable status of the trigger.

- given bullet "1" it's not quite clear what are expectations from, say an external debugger to perform a "clean"/"safe" reset. For example, your comments imply that this `Preset` state may be "a state that was before reset request. This means that if user requested some HW breakpoints to be set - then the debugger must ensure that those are cleared before reset. This is rather intrusive (in general case one may need to halt a hart in order to do this - and if a hart is in funky state, this may be problematic).

pdonahue-ventana, 2023-11-29:
The TM defines several CSRs and it also defines several types of triggers that are indirectly accessed via tselect+tdata1/2/3.  It defines the reset state of all of the CSRs plus all of the underlying triggers.

For instance, you may have an implementation with two mcontrol6 triggers and one etrigger trigger.  You will implement:
- tselect which resets to 0 according to https://github.com/riscv/riscv-debug-spec/blob/master/xml/hwbp_registers.xml#L33
- tcontrol which resets to 0 per https://github.com/riscv/riscv-debug-spec/blob/master/xml/hwbp_registers.xml#L191
- scontext which is reset to 0 per https://github.com/riscv/riscv-debug-spec/blob/master/xml/hwbp_registers.xml#L240
- mcontext which is reset to 0 per https://github.com/riscv/riscv-debug-spec/blob/master/xml/hwbp_registers.xml#L262 and which may also be accessible via hcontext
- trigger 0 which is accessible to software by writing tselect=0 and then doing indirect accesses via tdata1/2/3 and tinfo
  - trigger 0 contains a register that's accessible by writing tselect=0 and then accessing tdata1.  I'll call this register trigger0.tdata1.  It is in the format specified in mcontrol6.  trigger0.tdata1 has [XLEN-1:XLEN-4] hardwired to 6 and all the other bits are reset to 0 as specified starting at: https://github.com/riscv/riscv-debug-spec/blob/master/xml/hwbp_registers.xml#L714
  - trigger 0 contains a register that holds the address/data which is accessed by software by writing tselect=0 and then accessing tdata2.
  - trigger 0 contains a (optional) register that holds the trigger extra information which is accessed by software by writing tselect=0 and then accessing tdata3.
  - trigger 0 contains a (optional) register that holds the trigger info information which is accessed by software by writing tselect=0 and then accessing tinfo.  This is all hardwired to 0 except that [31:24] is hardwired to 1 and [6] is hardwired to 1.
- trigger 1 which is accessible to software by writing tselect=1 and then doing indirect accesses via tdata1/2/3 and tinfo
  - trigger 1 contains a register that's accessible by writing tselect=1 and then accessing tdata1.  I'll call this register trigger1.tdata1.  It is in the format specified in mcontrol6.  trigger1.tdata1 has [XLEN-1:XLEN-4] hardwired to 6 and all the other bits are reset to 0 as specified starting at: https://github.com/riscv/riscv-debug-spec/blob/master/xml/hwbp_registers.xml#L714
  - trigger 1 contains a register that holds the address/data which is accessed by software by writing tselect=1 and then accessing tdata2.
  - trigger 1 contains a (optional) register that holds the trigger extra information which is accessed by software by writing tselect=1 and then accessing tdata3.
  - trigger 1 contains a (optional) register that holds the trigger info information which is accessed by software by writing tselect=1 and then accessing tinfo.  This is all hardwired to 0 except that [31:24] is hardwired to 1 and [6] is hardwired to 1.
- trigger 2 which is accessible to software by writing tselect=2 and then doing indirect accesses via tdata1/2/3 and tinfo
  - trigger 2 contains a register that's accessible by writing tselect=2 and then accessing tdata1.  I'll call this register trigger2.tdata1.  It is in the format specified in etrigger.  trigger2.tdata1 has [XLEN-1:XLEN-4] hardwired to 5 and all the other bits are reset to 0 as specified starting at: https://github.com/riscv/riscv-debug-spec/blob/master/xml/hwbp_registers.xml#L1256
  - trigger 2 contains a register that holds the exception mask which is accessed by software by writing tselect=2 and then accessing tdata2.
  - trigger 2 contains a (optional) register that holds the trigger extra information which is accessed by software by writing tselect=2 and then accessing tdata3.
  - trigger 2 contains a (optional) register that holds the trigger info information which is accessed by software by writing tselect=2 and then accessing tinfo.  This is all hardwired to 0 except that [31:24] is hardwired to 1 and [5] is hardwired to 1.

To be clear, the total number of registers is 4 + N*4 where N is the number of triggers (4 because of tselect, tcontrol, mcontext, scontext and 4 because of the tdata1/2/3 and tinfo for each trigger).  In this example, there are 4 + 3*4 = 16 registers and all 16 have fully specified reset values.  If you implement 128 triggers then you will have 4 + 128*4 = 516 registers and all 516 will basically all reset to 0 except for the hardwired fields and icount.count in the tdata1 register of any icount triggers.  All 516 of these registers live in the trigger module.  The trigger module is part of the hart.

The number of registers is separate from the number of CSR addresses.  You can access the single mcontext register by using the mcontext CSR address or using the hcontext CSR address (multiple CSR addresses map to the same register).  You can access triggerX.tdata1 by setting tselect=X and then accessing tdata1 (multiple registers are accessible via the same tdata1 CSR address).

The reset behavior of all of the registers is fully specified in the XML and the spec.  If this is still unclear, perhaps @timsifive can provide more guidance about the purpose of the "Reset" column in each table.

You might also look at Smcsrind and Sscsrind for a similar mechanism where CSR addresses and underlying registers are independent.

aap-sc, 2023-11-29:
@pdonahue-ventana first of all, thank you for you reply. It really helps. However, I still have some concerns:

1. 
> It defines the reset state of all of the CSRs plus all of the underlying triggers.

That's my concern. I don't see this in the specification

2.

> To be clear, the total number of registers is 4 + N4 where N is the number of triggers (4 because of tselect, tcontrol, mcontext, scontext and 4 because of the tdata1/2/3 and tinfo for each [...]

I don't see this in the spec too. We have this "Table 5.3: Trigger Registers" . It enumerates all the registers:
[image]

I should say that the naming of the table is somewhat confusing, since I interpret it as "Trigger Module Registers"

3. tselect which resets to 0 according to `https://github.com/riscv/riscv-debug-spec/blob/master/xml/hwbp_registers.xml#L33 `. For whatever reason the **pdf** does not list "0" as a reset vector. Is it something that needs to be fixed?

4.

>  If this is still unclear, perhaps @timsifive can provide more guidance about the purpose of the "Reset" column in each table.

As I've said in the comment above, these reset values do not define **enabled/disabled** status of the trigger.

timsifive, 2023-12-01:
I see a slight different in your interpretation of things. @pdonahue-ventana is saying that all triggers have reset values defined. @aap-sc says the CSRs have reset values defined, but it's not guaranteed that the underlying triggers that are accessed through those are also reset.

The intent of the spec is for all triggers to be disabled after a reset that resets the Trigger Module. I'll
1. Add a line that says triggers are disabled after the TM is reset.
2. Rename Trigger Registers section to Trigger Module Registers.
3. Clarify that the reset values defined on tdata* and textra* apply to the underlying triggers.

Does that work?

aap-sc, 2023-12-01:
> @pdonahue-ventana is saying that all triggers have reset values defined. @aap-sc says the CSRs have reset values defined, but it's not guaranteed that the underlying triggers that are accessed through those are also reset.

yes, exactly.

> Does that work?

Awesome! Thanks!

aap-sc, 2023-12-05:
regarding this one:

> Add a line that says triggers are disabled after the TM is reset.

@timsifive  I just want to double-check if there are still plans to add this line.

timsifive, 2023-12-05:
Yes. I plan to write up something about reset today.

timsifive, 2023-12-05:
I'm adding that the reset values apply to the triggers. Given the reset values themselves, I don't think we need a line that says triggers are disabled by a reset. (I'm not opposed to adding it, but I couldn't find an obvious place where it fits without creating a new section, which seems a little overkill for this.)

aap-sc, 2023-12-05:
@timsifive this is a little confusing actually. 

> I'm adding that the reset values apply to the triggers. Given the reset values themselves, I don't think we need a line that says triggers are disabled by a reset.

Consider this:

1. we have only one value of `tselect` and `tdata*` active at a time. Consider tselect = 0. We have a fixed `tdata*` after reset. However, what about the second trigger? 
2. we don't have "enable" bit that can be read via `tdata` registers. The portable way to enable a trigger is to write non-zero value to `tdata*` and make sure that we read the same value back. To disable a trigger we just need to write "0" (the spec is silent regarding which value is read back in this case).

timsifive, 2023-12-05:
The new spec makes it clear that the CSRs access the triggers, and the reset values documented on the CSR apply to every trigger.

aap-sc, 2023-12-05:
@timsifive could you please take a look at my previous comment? 

> The new spec makes it clear that the CSRs access the triggers, and the reset values documented on the CSR apply to every trigger.

I'm not sure that this helps...

timsifive, 2023-12-05:
I don't think I quite understand what you're getting at.

> 1. we have only one value of `tselect` and `tdata*` active at a time. Consider tselect = 0. We have a fixed `tdata*` after reset. However, what about the second trigger?

For tdata* the spec now says "The reset values listed here apply to every underlying trigger." I think that clearly includes the second trigger.

> 2. we don't have "enable" bit that can be read via `tdata` registers. The portable way to enable a trigger is to write non-zero value to `tdata*` and make sure that we read the same value back. To [...]

Right. I don't understand how that is relevant here. As far as reset is concerned, the reset values always put m, s, h, vs, and vu to 0, which means the trigger is inactive.

aap-sc, 2023-12-06:
> For tdata* the spec now says "The reset values listed here apply to every underlying trigger." I think that clearly includes the second trigger.

Yes. It does apply. However, consider the following system: trigger 0 supports only address match, trigger 1 supports only icount matches. This means that they have different set of legal values for tdata1. However, we have only one reset value for tdata1. This is why I think this is confusing.

> Right. I don't understand how that is relevant here. As far as reset is concerned, the reset values always put m, s, h, vs, and vu to 0, which means the trigger is inactive.

Ouch! Re-read the spec. Yes, now I see.  "enable" bit is not an issue anymore. Thanks!