riscv/riscv-debug-spec #930: How to define the action of hcontext or textra32's action, when H extemsion is enabled or disabled by setting misa[7].

vang2333, 2023-12-20:
RISC-V Privileged Architecture:
The hypervisor extension is enabled by setting bit 7 in the misa CSR, which corresponds to the letter H. RISC-V harts that implement the hypervisor extension are encouraged not to hardwiremisa[7], so that the extension may be disabled.
RISC-V DEBUG Specification:
when H extension is not supported, the only  legal value of mhselect's value in textra register  are 0 and 4.  

So if we set mhselect = 1/2/5/6 when misa[7] = 1, what value should we read when we later set misa[7] = 0. 

If the H extension is implemented, it’s recommended to implement no more than 7 bits on RV32 and 14 on RV64.
Hcontext register is only accessible in S/HS-mode, M-mode and Debug Mode.
So, whether we can read/write the hcontext, when set misa[7] = 0, and what the length will be, when we change the misa[7].

pdonahue-ventana, 2023-12-20:
> So if we set mhselect = 1/2/5/6 when misa[7] = 1, what value should we read when we later set misa[7] = 0.

That's WARL so you can read back any value that's legal when misa.H=0.  It seems easiest to return `mhselect & {1'b1, misa.H, misa.H}` but, like all WARL fields, that's entirely up to the implementation.

> If the H extension is implemented, it’s recommended to implement no more than 7 bits on RV32 and 14 on RV64.\
> Hcontext register is only accessible in S/HS-mode, M-mode and Debug Mode.\
> So, whether [...]

If you implement 6 bits on RV32 or 13 bits on RV64 then you'll comply with the recommendation for both misa.H=0 and misa.H=1.  But limiting the width is just a recommendation.  I think that it's not allowed to have the width depend on misa since no width is illegal.

vang2333, 2024-03-15:
Resolved

closed 2024-03-15 by vang2333