riscv/riscv-debug-spec #957: Sdtrig: state of triggers when debugged process moves to another CPU

hschauhan, 2024-01-17:
Triggers are per-cpu. What happens when a trigger is set on an address of a process on one CPU and the process then is migrated to another CPU? Ideally OS should keep track of this and move the triggers to another CPU. What happens in cases where the triggers are being set by the kernel debugger? During that time all CPUs but one may be frozen. The debugger may or may not be running on the target CPU where triggers are to be configured. The target CPU might be frozen.

pdonahue-ventana, 2024-01-17:
> Triggers are per-cpu.

They're per-hart.  Note that [simultaneous multithreading](https://en.wikipedia.org/wiki/Simultaneous_multithreading) means that a single CPU core may have more than one hart.

> What happens when a trigger is set on an address of a process on one CPU and the process then is migrated to another CPU? Ideally OS should keep track of this and move the triggers to another CPU.

I think that you answered your own question.

> What happens in cases where the triggers are being set by the kernel debugger? During that time all CPUs but one may be frozen. The debugger may or may not be running on the target CPU where triggers are to be configured. The target CPU might be frozen.

I don't exactly understand your scenario.  If frozen means halted then there's an external debugger running and the external debugger can use either the program buffer or abstract CSR accesses to write to the triggers.  If frozen means something like powered down then you're debugging a hart but the debugger itself is running on some other hart that is turned off?  I don't see how you can debug anything in that scenario (regardless of how triggers work).

hschauhan, 2024-01-18:
> I don't exactly understand your scenario. If frozen means halted then there's an external debugger running and the external debugger can use either the program buffer or abstract CSR accesses to [...]

I am speaking strictly of native debugging and the scenario where kernel debugger is running. The other harts will spin and one hart runs to inspect the state of the system. In such case, when a trigger is set and after the kernel debugger resumes the process may migrate to another CPU. Like I answered my own question previously, OS can again do state management and make sure triggers move wherever the process moves.

What I am eventually coming towards is that if like DR0-DR7 x86 registers, a global and local debug registers make sense, making life of OS a bit simpler

pdonahue-ventana, 2024-01-19:
I'm not very familiar with the x86 debug architecture.  Based on a quick search, it sounds like these registers are per CPU (or maybe per thread if you have Hyperthreading) and are not global in the sense of being system-wide.  That makes sense because I'm not sure how you could seamlessly distribute the breakpoint information to all the CPUs in a multi-chip system.  As far as I can tell from my quick (and probably inadequate) research, the terms "global" and "local" seem to mean that the breakpoint applies to all processes or it applies to a single process.

This seems similar to RISC-V's textra/tdata3 register.  You can associate a trigger with a specific context/ASID/VMID (local) or not (global).  With this textra filtering and/or the mcontrol6 mode filtering, the debugger can program triggers on every hart and those triggers will only be active when running the particular user process in question (U mode and scontext/ASID), the particular guest OS (VS mode and hcontext/VMID), or guest user process (VU mode and hcontext/VMID and scontext/ASID) or when running the hypervisor/OS (S/HS mode and maybe mcontext).

ved-rivos, 2024-01-19:
> the terms "global" and "local" seem to mean that the breakpoint applies to all processes or it applies to a single process.

Yes. The local and global enables in DR7 were designed for hardware managed task switching. Hardware task management is not used by modern 32-bit OS and the x86-64 ISA does not support it either.

pdonahue-ventana, 2024-01-25:
I think that the questions are answered.  You can make a follow-up comment if not.

closed 2024-01-25 by pdonahue-ventana