riscv/riscv-debug-spec #992: Register bit-field bit offset numbers clipped

TommyMurphyTM1234, 2024-03-23:
Minor/cosmetic issue.

Is it possible to fix the clipping of the top of the bit-field bit offset numbers here (and elsewhere)?

[image]

rtwfroody, 2024-03-26:
This seems like a PDF rendering/font issue. Looking at the rc2 PDF in my browser, I get:
[image]

We're not requesting a specific font, or anything. :( The relevant code is at https://github.com/riscv/riscv-debug-spec/blob/a054e5684a40397404b8094ba99f93c5e053cb7b/registers.py#L927

I don't know how to fix this.

pdonahue-ventana, 2025-02-13:
I assume that this is not still a problem.  Reopen if it is.

closed 2025-02-13 by pdonahue-ventana