riscv/riscv-debug-spec #994: Typo on the minimmum requirements of Debug Mode

albert-azcarate, 2024-03-25:
On chapter 3, when specifing the mandatory things to implement, on point 3.C:
"Implement abstract access to at least all GPRs, [descr], and [dpc], and advertise the implementation as conforming to the "Minimal RISC-V Debug Specification", instead of the "RISC-V Debug Specification"."

[descr] should be dcsr.

pdonahue-ventana, 2024-03-25:
In addition to that problem, it looks like the hyperlinks don't work.  @rtwfroody: Should `<<x>>` be `{x}` or does that not work across chapters?

albert-azcarate, 2024-03-26:
On chapter 3.2:
"The havereset bits for the selected harts can be cleared by writing 1 to {dmstatus-ackhavereset} in dmcontrol. The havereset bits might or might not be cleared when dmactive is low."
 {dmstatus-ackhavereset}  should hyperlink  to the ackhavereset on the dmcontrol definition and should only display "ackhavereset".

albert-azcarate, 2024-03-26:
Broken hyperlinks with [ ]:
Chapter 3.9: "the different fields of dmcontrol, abstractcs, [abstractauto], and command."
1.2.1.3. Minor Changes from 0.13 to 1.0:  
    "4. All Trigger Module registers ([tab:trigger]) are optional"
    "14. Change quick access exceptions to halt the target in [Quickaccess]."

rtwfroody, 2024-03-29:
I've fixed all these in #997. @albert-azcarate thanks for pointing out these problems.

closed 2024-04-24 by albert-azcarate