riscv/riscv-debug-spec #995: Q about Table 9 (action encoding)

martinberger, 2024-03-27:
I have a question about Table 9 (action encoding) in Section 5.2 on Page 59 in 1.0.0-rc2. In in the row describing the action to be taken if the action is 1 we have the sentence

>    *"hardware should clear the action field whenever the action field is  1, the new value of dmode would be 0, and the new value of he action    field would be 1."*

It is unclear because the sentences asks to *"clear"* the action field but also asks to set it to 1. This clearing to 1 should only happen if the action field is 1.

(There is also a typo: *"**he** action field"* should the *"**the** action field"*.)

pdonahue-ventana, 2024-03-27:
IF the current action field is 1 and hardware is doing a write that would write (1) dmode=0 and (2) action=1 THEN hardware should write action=0 (rather than allowing item 2).

tomaird, 2024-04-04:
From your answers to previous issues (https://github.com/riscv/riscv-debug-spec/issues/746 and https://github.com/riscv/riscv-debug-spec/issues/749), it seems the intention is that it's impossible to get in the state where `dmode=0` and `action=1`, in which case I think the spec wording is unclear.

I propose this sentence: 

> Since the tdata registers are WARL, hardware should clear the action field whenever the action field is 1, the new value of dmode would be 0, and the new value of he action field would be 1.

Should be changed to this:

> Since the tdata registers are WARL, hardware should clear the action field whenever the new value of dmode would be 0, and the new value of the action field would be 1.

Since it doesn't actually matter what the _current_ value of `action` is, it only matters what the _resulting_ combination of `action` and `dmode` are

pdonahue-ventana, 2024-04-04:
Yes, it's illegal to have dmode=0 and action=1 and the WARL logic must prevent that situation.  The spec provides a recommendation for one specific situation (see https://github.com/riscv/riscv-debug-spec/pull/501 for some history).

A debugger can attempt to write dmode=0 and action=1 in multiple scenarios:
 1. The previous state had action=1 and dmode=1.
     - The current spec directly addresses this.
     - This might be a buggy external debugger (but not a buggy native debugger).
     - The WARL hardware could either:
       - write dmode=1 and action=1 (dropping the write)
       - write dmode=0 and something fixed like action=0 (which is the current spec's recommendation)
 2. The previous state had action!=1 and dmode=1.
     - This might be a buggy external debugger (but not a buggy native debugger).
     - The WARL hardware could either:
       - write dmode=1 and action=1 (dropping the write to dmode)
       - write dmode=0 and something fixed like action=0 (which is your proposal)
       - write dmode=0 and keep the original value of action
 3. The previous state had action!=1 and dmode=0.
     - This might be a buggy native or external debugger.
     - The WARL hardware must write dmode=0 (which was both the previous and new value)
     - The WARL hardware could either:
       - write something fixed like action=0 (which is your proposal)
       - keep the original value of action

I think that the spec didn't want to pick one of the options on scenarios 2 and 3.  I don't really know why.

tomaird, 2024-04-05:
I don't think it's clear enough that setting `action=1` and `dmode=0` is a mandatory restriction but the example way of dealing with it is only a suggestion. The two flow together to imply that both are restrictions.

I also think it's quite confusing for the spec to provide a suggestion for 1 scenario and not for the other 2.

It would be better to either:

1. Provide a suggestion for 1 scenario, but make it clear that it's a suggestion and also that there are other scenarios to consider
2. Provide a suggestion for all 3 scenarios, again making it clear that they are suggestions.
3. Provide no suggestions, only what is actually restricted, and leave it entirely up to each implementation to decide how to handle all the scenarios

pdonahue-ventana, 2024-04-05:
I'd personally choose 3 and say something like "Since the tdata registers are WARL, hardware must prevent them from containing dmode=0 and action=1."  @rtwfroody: You came up with the original language.  Thoughts?

rtwfroody, 2024-04-12:
I agree that the right thing is to document the invalid value, and leave it up to the hardware what valid value it ends up with.

> Since the tdata registers are WARL, hardware must prevent them from containing dmode=0 and action=1.

If we say must, then that's not backwards-compatible. "Should" is backwards-compatible.

Do we need ARC review to make the change using "should?"

pdonahue-ventana, 2024-04-15:
There was always a requirement that hardware must prevent dmode=0 and action=1.  "This action is only legal when the trigger's dmode is 1" means that dmode=0 and action=1 is illegal.  WARL says that "If a value is unsupported, the implementation converts the value to one that is supported."  Previously, there was a recommendation about the specific supported value that hardware would convert to in one particular case.  My suggestion is to remove that and just specify the requirement.

We could remove that WARL sentence altogether and just say "This action is only legal when the trigger's dmode is 1" but apparently people didn't understand that the WARL hardware needs to prevent the illegal situation.

pdonahue-ventana, 2025-04-05:
I believe that this was addressed in #1009.  Reopen if not.

closed 2025-04-05 by pdonahue-ventana