riscv/riscv-debug-spec #999: NAPOT matching

tomaird, 2024-04-08:
Can someone clarify this section of the spec for NAPOT matching in `mcontrol6`:

> Matches when the top M bits of any compare value match the top M bits of tdata2. M is XLEN-1 minus the index of the least-significant bit containing 0 in tdata2

If `tdata2 = 0x000000fffffffff4` and we execute an instruction with PC `0x000000fffffffff0`, should this match?

Does the above quote mean:

1. The least significant bit containing 0 in `tdata2` is bit 0. Bit 0 is the lowest index bit that is zero. So `M = XLEN-1`

Or does it really mean: 

2. The MSB index of the LSBs containing continuous 0s, which would be bit 1. i.e. `M=XLEN-2`

Or is it:

3. Something else...

pdonahue-ventana, 2024-04-15:
It literally means the least significant bit containing 0, which is the first thing you said.  This matches the NAPOT definition in PMP.  See the table in the priv spec.

The only two PCs in your example that would match would be 0x000000fffffffff4 and 0x000000fffffffff5.  Because pc[0]=0, the second one is impossible so it would be more straightforward for the debugger to just use match=0 in this case.

rtwfroody, 2024-04-24:
I assume Paul's answer is sufficient here.

closed 2024-04-24 by rtwfroody