riscv/riscv-debug-spec #1008: Does *stateen0[59] control mscontext?

YenHaoChen, 2024-04-17:
Hello, 

The *stateen0[59] controls scontext. When mstateen0[59]=0, accessing scontext from S/HS-mode raises an exception. Does the *stateen0[59] also control the mscontext, the alias of scontext, if implemented?

[image]

pdonahue-ventana, 2024-04-17:
mscontext is only for Debug 0.13.  mstateen0 is part of Smstateen.  The assumption probably was that nobody would implement both the old debug spec and the relatively new Smstateen extension.

Since the purpose of Smstateen is to prevent access to certain state rather than certain CSRs, I think that mstateen0[59] should control access to mscontext in this unusual implementation.

YenHaoChen, 2024-04-17:
Thank you for the clarification.

closed 2024-04-17 by YenHaoChen