riscv/riscv-debug-spec #1011: Timing requirements for a chain of `mcontrol` load address triggers

en-sc, 2024-04-24:
[5.7.11. Match Control (mcontrol, at 0x7a1)] states:

> A chain of triggers that don’t all have the same timing value will never fire. That means to implement
the suggestions in Table 11, both timings should be supported on load address triggers.

Is this correct? Seems like it's enough to support `timing == before (0)` since [Table 11. Suggested Trigger Timings] recommends `timing == before` for Load Address triggers.

rtwfroody, 2024-04-25:
To implement the suggested timings, load address triggers need:
1. timing=before, to implement a simple load address before
2. timing=after, to implement load address + data after

Which means that load address triggers need both before and after timing support. Does that make sense?

en-sc, 2024-04-26:
Oh, I see, I have misunderstood.
So `That means to implement the suggestions in Table 11...` implies **all** suggestions from the table.
Maybe something like:
> That means to implement the suggestions in Table 11, both timings should be supported on a load address trigger if it can be chained with a load data trigger.

would be more clear?