riscv/riscv-debug-spec #1013: `mcontrol' / 'mcontrol6` `match=lt` trigger on the last valid address

en-sc, 2024-04-24:
AFAIU, this is not possible to setup a `mcontrol` / `mcontrol6` trigger with `match=lt` that would match on the access to the last valid address, since `tdata2` would need to contain at least `<last valid address> + 1`, which is invalid and may not be representable by `tdata2` ([5.5.4. Address Matches])
> For address matches without a mask, tdata2 must be able to hold all valid addresses in all supported translation modes. That means that after writing any of these valid addresses, the exact same value XLEN-wide value is read back, including any high bits. An implementation may be able to optimize the storage required, depending on the widest addresses it supports.

This may become an issue if a user wishes to place a multi-byte unaligned watchpoint that includes the last valid address.
Such a watchpoint can not be reliably placed with `match=eq` triggers, since there is no way to infer how many comparison values there are. So the debugger will try to setup it by using `lt`-`ge` chain, but it will fail because the desired `tdata2` value is not representable.

pdonahue-ventana, 2024-04-24:
> AFAIU, this is not possible to setup a `mcontrol` / `mcontrol6` trigger with `match=lt` that would match on the access to the last valid address, since `tdata2` would need to contain at least `<last [...]

This would be implementation-dependent.  The fact that `tdata2` must be able to hold all valid addresses doesn't mean that it cannot also hold `<last valid address> + 1` if the implementer wants to.

There are multiple types of implementations:
- those that won't support what you want because `tdata2` can't hold `<last valid address> + 1`
- those that won't support what you want because they don't support `match=lt`
- those that won't support what you want for both reasons
- those that will support what you want

For better or worse, almost everything in Sdtrig is optional.  Implementers need to consider the tradeoffs (including the possibility that a strange or minimalist implementation would not be supported by debuggers).

> This may become an issue if a user wishes to place a multi-byte unaligned watchpoint that includes the last valid address.

How about a `match=napot` trigger on the last few bytes of memory?  (Of course, `match=napot` is optional like everything else but at least you know that `tdata2` will be able to hold the necessary value.)

rtwfroody, 2024-04-25:
> This may become an issue if a user wishes to place a multi-byte unaligned watchpoint that includes the last valid address.

Can't you simply set a ge trigger for this case? Why do you need to chain it with an lt trigger?

en-sc, 2024-04-26:
> Can't you simply set a ge trigger for this case? Why do you need to chain it with an lt trigger?

But how a debugger is expected to figure out the last valid address?

pdonahue-ventana, 2024-04-26:
> But how a debugger is expected to figure out the last valid address?

For Sv57/48/39/32, it's defined by those sections in the priv spec (and doesn't vary between implementations).  That's the easy part.

For Bare mode or M mode, it's the last address that's not a vacant PMA region.  The debugger needs to figure it out in the same way that M mode software figures it out: by reading the configuration structure.  M mode uses mconfigptr for that and the debugger uses confstrptr.

closed 2024-04-26 by en-sc