riscv/riscv-debug-spec #1014: `maskmax6` discovery may need clarification

en-sc, 2024-04-24:
The specification describes the following algorithm (referenced bellow as algorithm (1)) to discover the value of `maskmax6`:

> The value of maskmax6 can be determined by the debugger via the following sequence:
> 1. Set match=1.
> 2. Read match. If it is not 1 then NAPOT matching is not supported.
> 3. Write all ones to tdata2.
> 4. Read tdata2. The value of maskmax6 is the index of the most significant 0 bit plus 1.

AFAIU, this contradicts a bit with an algorithm for setting up a trigger [5.7. Trigger Module Registers]:

> As a result, a debugger can write any supported trigger as follows:
> 1. Write 0 to tdata1. (This will result in containing a non-zero value, since the register is WARL.)
> 2. Write desired values to tdata2 and tdata3.
> 3. Write desired value to tdata1.

My understanding is, the HW may implement WARL policies for `tdata*` as follows:
Write to any `tdata*` register of a value that is illegal for the current configuration results in this register retaining the previous value (the one before the write).

This will result in a failure of the algorithm (1), since step 1 involves changing `tdata1` while `tdata2` may contain an illegal value for the new `tdata1` value. So step 2 will result in a false negative (NAPOT is supported but `match` is not equal to 1).

en-sc, 2024-04-24:
AFAIU, this can be fixed by modifying algorithm (1) by adding tow steps in the beginning:
1. Write `tdata1=0` (to be able to write `tdata2`)
2. Write `tdata2=0` (AFAIU, zero is always valid for NAPOT match, regardless of `maskmax6`)

rtwfroody, 2024-04-25:
I think you are right. I'll make a PR.

closed 2024-06-05 by en-sc