riscv/riscv-debug-spec #1016: How to halt or resume multiple harts by DM in one DM-one Hart SoC?

zhangdujiao, 2024-04-25:
For a single Debug Module (DM) corresponds to a single hart System-on-Chip (SoC) system, how can multiple harts be simultaneously halted or resumed synchronously?
Thanks!

en-sc, 2024-04-25:
https://openocd.org/doc-release/html/Config-File-Guidelines.html
Section [6.3.4 Define CPU targets working in SMP]

en-sc, 2024-04-25:
AFAIU, the debug spec does not cover this. The functionality you need may be supported by a custom mechanism. One should also be mindful whether the DMs share a single TAP (more precisely, DMs are on the same DMI bus) or not.

zhangdujiao, 2024-04-25:
> One should also be mindful whether the DMs share a single TAP (more precisely, DMs are on the same DMI bus) or not.

thanks. 
yes,  assuming all DMs are on the same DMI bus.

rtwfroody, 2024-04-25:
The debug spec doesn't say. You could probably rig something up with halt groups, resume groups, and external triggers. In this case the external triggers would somehow connect to the harts on other DMs. Section 3.6 talks about the basic mechanisms.

closed 2024-06-12 by zhangdujiao