riscv/riscv-debug-spec #1025: Cross Trigger definition in RISC-V

zhangdujiao, 2024-04-29:
Since RV doesn't define the cross trigger, can we just consider the external trigger as cross trigger? But what is the type of external trigger, I didn't see in the tmexttrigger.select.
But we can find 2 types of external trigger, which are machine mode performance counter overflow and interrupts in 3.3.2. External triggers RISC-V External Debug Security Extension. Where can we find other types of external trigger?

Many Thanks!

pdonahue-ventana, 2024-04-29:
> Since RV doesn't define the cross trigger, can we just consider the external trigger as cross trigger?

This is all implementation-specific but here's the idea:
- If you have multiple harts connected to a single DM then halt groups and resume groups are the way to cross trigger.  If one hart halts then it tells the DM (so that the DM can update things like anyhalted/allhalted).  Assuming that you've programmed your halt groups properly, the DM then tells the other harts in that halt group to halt.  And similar on resume.
- If you have multiple harts with one hart per DM then dmeexttrigger is the way to cross trigger.  When one hart halts, it tells its DM as usual.  That DM should be programmed to fire the output dmexttrigger.  That output should be connected to the other dmexttrigger input on the other DM(s).  The other DM(s) should be programmed to cause the hart to halt when the dmexttrigger input fires.
- If you have multiple DMs and each has multiple harts then you combine the two approaches.  The harts that are connected to the same DM as the halting hart will use halt groups to be cross triggered and the harts that are connected to other DMs will use dmexttrigger to be cross triggered.

You could also use tmexttrigger in certain cases like maybe if you have a SMT core, but in most other cases this would involve routing more wires around than the above scheme.

zhangdujiao, 2024-04-30:
Thank you very much for your patience in answering my questions. I have a better understanding of the role of dmexttrigger in halt and resume groups, but I still have a few questions.
We want to figure out the types of cross triggers required for an RV core. Referring to the ARM implementation, I know that it generally includes debug/halt request, resume request, trigger from/to interrupt controller, trigger from/to ETM (encoder), etc., a total of 10 cross triggers.
As you mentioned, debug/halt request and resume request can be used as cross triggers for halting and resuming groups in a RV hart. Besides, are there other types of triggers that need to be implemented in a RV hart? For example, trace on, trace off, trace notify, etc., which are introduced in the trace specification.

closed 2024-06-12 by zhangdujiao