riscv/riscv-debug-spec #1029: Ambiguous requirement about dmcontrol behavior when dmcontrol.dmactive==0

JanMatCodasip, 2024-05-13:
The debug spec says the following about the `dmcontrol` register when `dmcontrol.dmactive==0`:

> 0 (inactive): The module’s state, including authentication mechanism,
> takes its reset values (the dmactive bit is the only bit which can be written to something other than its reset value).

I believe the sentence `the dmactive bit is the only bit which can be written to something other than its reset value` can be interpreted in two ways - as a requirement for the external debugger vs. requirement for the debug module (DM):

- 1) The external debugger must not write any bit of dmcontrol, apart from dmactive, to any other value than the reset value.
- 2) The DM must not allow any bit of dmcontrol, apart from dmactive, to be written to a non-reset value by the external debugger. That is, all bits of dmcontol (except for dmactive) are read only and have the reset value.

I tend to believe the intention of the spec was the variant 2), but please correct me if that is incorrect assumption.

In any case, I would recommend to clarify this sentence in a future version of the debug spec.

pdonahue-ventana, 2024-05-13:
Everything in the DM is held in reset except for the dmactive bit itself.  Because the DM is in reset, it must not allow any state to contain the non-reset value (which is your second variant).  The very next sentence says "Any accesses to the module may fail" which means that the external debugger shouldn't be doing any writes at all (of any values, including the reset values).

JanMatCodasip, 2024-05-14:
@pdonahue-ventana, thank you for the clarification.

closed 2024-05-14 by JanMatCodasip