riscv/riscv-debug-spec #1044: Can tselect be written with an invalid index

jrahmeh, 2024-06-10:
Section 5.7.1 of the debug spec (version 1.0) states:

> To verify that what they wrote is a valid index, debuggers can read back the value and check that tselect holds what they wrote and read tdata1 to see that type is non-zero.

This seems to imply that if an attempt is made to write an invalid index into TSELECT, the actual value written will be different that what was attempted and it will be a valid index.  Am I reading this correctly, or are implementation allowed to write an invalid index into TSELECT?

pdonahue-ventana, 2024-06-10:
tselect says:
> This register is WARL. 

The definition of WARL is:
> Write any, read legal. A debugger may write any value. If a value is unsupported, the implementation converts the value to one that is supported.

So the actual value written might be different than what was attempted.  As a concrete example, if you only have 4 triggers then you only have to implement the low 2b of tselect rather than implementing all 32b or 64b.  If you only have 3 triggers then you can implement 2b but the value 3 must map to one of the 3 legal values.

This is exactly like how WARL works in the privileged architecture.

jrahmeh, 2024-06-10:
Thank you Paul Donahue.

closed 2024-06-10 by jrahmeh