riscv/riscv-debug-spec #1049: Are the vs and vu bits in mcontrol6 dependent on misa.h?

ThimayaPR, 2024-06-19:
Spec 5.7.12 about the tdata1.vu bit: 

> When set, enable this trigger in VU-mode. This bit is hard-wired to 0 if the hart does not support virtualization mode.

Does this imply the bit is writeable if the hart is capable of virtualisation regardless of the misa.h setting?

rtwfroody, 2024-06-19:
This text was probably written without considering that misa.h is writable. The answer hinges on what it means to support a virtualization mode. Does the ISA spec define that term?

To me it makes most sense to consider whether the feature is implemented, regardless of whether it is currently enabled. Further support for that idea is that the intro to section 5.7 says "Almost all trigger functionality is optional" and does not explicitly mention that the supported features may depend on other hart state.

For the debugger it would be simpler to always allow setting vu regardless of misa, since it makes discovery easier. Changing what triggers can be set depending on misa is just more to keep track of.

rtwfroody, 2024-06-19:
@pdonahue-ventana mentioned in email:
> Take a look at the definition of misa.  If misa.h=0 then it's supposed to behave exactly as if H isn't implemented at all.  That means that vu/vs should be read-only 0.

closed 2024-06-30 by ThimayaPR