riscv/riscv-debug-spec #1050: Question about mcontrol6.match and tdata2

micreven, 2024-06-27:
Hi All，
When I read the debug SPEC about trigger function. I see the SPEC said **"Legal values for tdata2 require M + maskmax6 ≥ XLEN and M > 0. See above for how to determine maskmax6."**  at page82 as below :
[image]

If I implement a hardware which "maskmax6"=64, the value 64'h7FFF_FFFF_FFFF_FFFF seems become a illegal value. Because the M will equal 0 (63-XLEN-1). And as the flow of getting maskmax6 value, when I write 64‘hFFFF_FFFF_FFFF_FFFF, I must read back 64'h7FFF_FFFF_FFFF_FFFF which indicates the maskmax6 value for debugger. 

And the SPEC said the tdata2 is WARL attribute, the 64'h7FFF_FFFF_FFFF_FFFF is a illegal value but must read back this illegal value for maskmax6  getting flow which seems contradiction.

Any help would be greatly appreciated!

Best regards,
Micreven

pdonahue-ventana, 2024-06-27:
I don't think that maskmax6 is allowed to be greater than 63.  If maskmax6=63 and tdata2 is programmed to have the largest allowed mask then we ignore the low 63 bits and compare the upper 1 bit.  What would it mean to to have maskmax6=64 and the value 0x7FFFFFFFFFFFFFFF which ignores the low 64 bits and compares the upper 0 bits?

micreven, 2024-06-28:
> I don't think that maskmax6 is allowed to be greater than 63. If maskmax6=63 and tdata2 is programmed to have the largest allowed mask then we ignore the low 63 bits and compare the upper 1 bit. [...]

Hi Paul，

thanks for your explanation.
But I can't found any limitation description for maskmax6 on the SPEC.
Is it a implicit limitation ？（maskmax6 must <= 63)

thanks
Micreven

pdonahue-ventana, 2024-06-28:
I think that maskmax6<=63 for the reasons you pose in the original question.  Specifically, because M>0.

Imagine that we remove the M>0 requirement and say that maskmax6 is allowed to be 64.  Then tdata2=0x7FFFFFFFFFFFFFFF is legal and that value will match on every single access.  That doesn't seem useful.  That's why the M>0 requirement was added.  That M>0 requirement means that at least one bit must be compared so the maximum mask size cannot mask out all 64 bits (i.e. maskmax6 cannot be 64).

micreven, 2024-06-28:
Got it. 
Thanks

pdonahue-ventana, 2025-04-05:
This seems to have been answered.

closed 2025-04-05 by pdonahue-ventana