riscv/riscv-debug-spec #1057: Question about Trigger Match Control Type6 CSR's "match" field function

micreven, 2024-08-03:
Hi All, 
 
I have a question about the mc6 trigger CSR's match field.

When I read the "match" field function, I think the compare formula should be like below if I config match=5.
"**results=((compare_value[63:32]&tdata2[63:32])==(tdata2[31:0]))**"

But the SPIKE code shows the tdata2[31:0] also ANDed with tdata[63:32], which seems mismatch with SPEC（Because I think the SPEC doesn't say the tdata2[31:0] should ANDed with tdata2[63:32]）

the SPIKE code copied below，from https://github.com/riscv-software-src/riscv-isa-sim/blob/master/riscv/triggers.cc

[image]

the same question is also existed for match=4、5、8、9.

Could anyone help me explain the SPEC description ? 
 
Best regards，
Micreven

pdonahue-ventana, 2024-08-06:
You're right.  The Spike code looks wrong.  It seems like it should be something like `((value >> (xlen/2)) & mask) == (tdata2 & ((1<<(xlen/2))-1)`.

If I want to match when value[1:0]=3 then I need to program tdata2=0x0000000300000003 where [31:2] must be 0 according to the spec.  If I have non-zero in [31:2] like 0x00000003DEADBEEF then the trigger cannot possibly match.  The Spike bug was probably never caught because nobody programmed the trigger in this way that made it impossible to match.

micreven, 2024-08-07:
Got it.
Thanks a lot.

closed 2025-02-13 by pdonahue-ventana