riscv/riscv-debug-spec #1058: Question about Trigger Match Control Type6 CSR's "size" field function

micreven, 2024-08-13:
Hi，All

I have a question about the mc6 trigger CSR's “size” field.

When I config the mc6.size=1, select=1, load=1, which means I want to trigger a load instruction whose access data equals the tdata2[7:0]. 

And my question is if a load instruction whose access size is 16bits but the low 8bits in the load data equals tdata2[7:0], Will this instruction match ?

Actually, I'm not sure about the size field for weather it is only used for indicating the compare data width for the execute/load/store data and tdata2 or it is also used to filter the real valid execute/load/store data to compare. 

Another example, If I set mc6.size=2, select=0, execute=1, Could this configuration filter out the 32bits length instruction ?

Any help would be greatly appreciated！

Best regards，
Micreven

rtwfroody, 2024-08-13:
> When I config the mc6.size=1, select=1, load=1, which means I want to trigger a load instruction whose access data equals the tdata2[7:0].
> 
> And my question is if a load instruction whose access size is 16bits but the low 8bits in the load data equals tdata2[7:0], Will this instruction match ?

No. For size=1 the spec says "The trigger will only match against 8-bit memory accesses."

> Actually, I'm not sure about the size field for weather it is only used for indicating the compare data width for the execute/load/store data and tdata2 or it is also used to filter the real valid execute/load/store data to compare.

The size field filters accesses only. It does not affect the compare value. That is what the select field is for.

> Another example, If I set mc6.size=2, select=0, execute=1, Could this configuration filter out the 32bits length instruction ?

For size=2, the spec says "The trigger will only match against 16-bit memory accesses or execution of 16-bit instructions." So that trigger would only fire when a 16-bit instruction is being executed (and other conditions specified by the match field etc. are satisfied).

micreven, 2024-08-14:
Got it.
Thanks a lot.