riscv/riscv-debug-spec #1063: Questions about DMI register op busy

zhangdujiao, 2024-08-21:
When reading the DMI register (scanning in a value with op==0 to shift out data in the DMI register), if the previous read/write operation to DM has not been completed, will it cause op to busy in capture-dr?  Does it mean that as long as the previous read/write operation to DM is not completed, the op will be 3 in capture-dr, regardless of the op of the value that will be scan in DMI register?

pdonahue-ventana, 2024-08-21:
Yes and yes.  When it does the capture, it can't possibly know what you're going to later scan into DMI.

See https://lists.riscv.org/g/tech-debug/topic/107145817#msg1697 for a recent discussion about this.

zhangdujiao, 2024-08-21:
many thanks :)

closed 2024-08-21 by zhangdujiao