riscv/riscv-debug-spec #1074: DRET executed while DCSR.STEP is set

jrahmeh, 2024-10-08:
If the DRET instruction is executed while DCSR.STEP is set, do we then re-enter debug mode?  Should DRET clear the DCSR.STEP bit?

JanMatCodasip, 2024-10-11:
Hello @jrahmeh,

My interpretation of the spec is that `DRET` (encoded as 0x7b200073) is not a valid RISC-V instruction. If it is found in a program, or if the external debugger tries to execute it via the program buffer, the outcome shall be an illegal instruction exception. The same should be true if the DRET is encountered during a single-step.

Some RISC-V debug implementations may choose that the debug module uses DRET (pushes it into the CPU's pipeline) to request exit from the debug mode, but that is an internal implementation decision. The above should still be observed.

Please see also [this commit](https://github.com/riscv/riscv-debug-spec/commit/822f63bdce09b9cd2f88eecf225eeb30daf6fd35#diff-65d0c637d2cfce68cfdce475fe4cff9e81d7d3cba188285d9e4de55cda88fe1eR67) which removed the DRET encoding from the main body of the spec and moved it to a non-normative appendix (just as one possible implementation choice for exiting the debug mode).

pdonahue-ventana, 2024-10-14:
@JanMatCodasip is exactly right.  Searching for "dret" shows that it appears only in one paragraph in a non-normative appendix.  Encoding 0x7b200073 is architecturally a reserved instruction encoding so any execution of 0x7b200073 outside of Debug Mode must behave like any other reserved encoding.

closed 2024-10-14 by pdonahue-ventana

jrahmeh, 2024-10-15:
Thank you @JanMatCodasip and @pdonahue-ventana.