riscv/riscv-debug-spec #1079: Does DCSR.mprven affect Abstract Access Memory commands?

aap-sc, 2024-10-28:
Here is the current description of `DCSR.mprven`

```
All operations are executed with machine mode privilege, except that additional Debug Mode CSRs
are accessible and mprv in mstatus may be ignored according to mprven. Full permission checks,
or a relaxed set of permission checks, will apply according to relaxedpriv.
```

Now here is roi from `3.7.1.3. Access Memory`

```
aamvirtual

0 (physical): Addresses are physical (to the hart they are performed on).
1 (virtual): Addresses are virtual, and translated the way they would be
from M-mode, with MPRV set.

Debug Modules on systems without address translation (i.e. virtual
addresses equal physical) may optionally allow aamvirtual set to 1, which
would produce the same result as that same abstract command with
aamvirtual cleared.
```

This creates a confusion since it's not quite clear whether DCSR.mprven should affect Abstract Memory Command or not. To me it seems that it should not. Moreover, it looks like DCSR.mprven is intended to be usable for program-buffer-like operations.

It would be nice to somehow clarify this in the specification. For example we could explicitly state if DCSR.mprven affect abstract memory access commands.

pdonahue-ventana, 2024-10-28:
The first thing you quoted is the second bullet under: "When executing code due to an abstract command, the hart stays in Debug Mode and the following apply:"  Only program buffer operations are "executing code due to an abstract command" so nothing in that list applies to abstract memory commands.

aap-sc, 2024-10-28:
> The first thing you quoted is the second bullet under: "When executing code due to an abstract command, the hart stays in Debug Mode and the following apply:" Only program buffer operations are "executing code due to an abstract command" so nothing in that list applies to abstract memory commands.

Got it. Thanks for clarification!

Do you think it is reasonable to try adjusting the text to highlight this distinction?  To me the difference between "abstract command" and "abstract memory command" is quite easy to miss.  The spec does not specify how "abstract memory commands" are implemented, and "abstract memory command" may lead to execution of some code under the hood.

vd-sc, 2024-10-29:
Let me jump in to the discussion. IMHO, the following statement

> Only program buffer operations are "executing code due to an abstract command" so nothing in that list applies to abstract memory commands.

seems to be a completely wrong interpretation of the Debug Spec.

Access Memory Command is just one of types of Abstract Commands (of type=2), same as Register Access Command (type=0), so DCSR influences both of them if they are executed in the Debug Mode. There are no signs in the spec to interpret Access Memory Command as a specific entity with different behavior in respect to Debug Mode and DCSR. Oppositely, statement that DCSR determines all Debug Mode behavior is more natural and seems harmonic with other architecture approaches (including other details of the Debug Mode behavior in the section 4.1).