riscv/riscv-debug-spec #1089: How Native debugger

zhangdujiao, 2024-12-02:


pdonahue-ventana, 2024-12-02:
The spec's description of dmcontrol.haltreq says:
> Writing 1 sets the halt request bit for all currently selected harts. Running harts will halt whenever their halt request bit is set.

To halt multiple harts, you select the multiple harts that you want to halt and then write 1 to haltreq.

zhangdujiao, 2024-12-02:
> > Writing 1 sets the halt request bit for all currently selected harts. Running harts will halt whenever their halt request bit is set.

thanks, it can halt multiple hart by access dmcontrol of DM. While from the information I received from the spec, native debugger to debug a program by raising a breakpoint execption rather than by accessing the DM.
The breakpoint execption is programmed by CSRs hardware breakpoint registers as shown in "5.7. Trigger Module Registers
". So, can native debugger access the trigger module CSRs of all harts?

pdonahue-ventana, 2024-12-02:
I missed the word "native" in your question.  Native debuggers cannot halt harts at all.  Halting involves entering Debug Mode, where the hart is controlled by an external debugger (e.g. doing abstract reads/writes, executing from the program buffer, resuming).

A native debugger just deals with traps.  If you want to cause multiple harts to trap asynchronously, you probably should send them interrupts.  Or you could maybe use tmexttrigger in an implementation-defined way.

zhangdujiao, 2024-12-02:
> If you want to cause multiple harts to trap asynchronously, you probably should send them interrupts. Or you could maybe use tmexttrigger in an implementation-defined way.

So, the native debugger can **only debug the program running on the core itself** by configure the CSRs of the Trigger Module.
Once the trigger register fires, the debugger can cause other harts entering the traps by sending interrupts or tmexttrigger. What is the purpose of causing other harts to enter trap? The other harts may not enter the trap at the same time with the hart with debugger since the latency from the link。

zhangdujiao, 2024-12-03:
We are testing on a multi-core RISC-V SoC from SOPHGO, where GDB running on CPU0 can debug any other CPUs. Including stopping program, setting breakpoints, and single-step. This seems different from the RV debug spec, which will debug the core by accessing the Trigger Module CSRs. From the spec definition, GDB running on CPU0 cannot debug other CPUs since the CSRs registers are not shareable fro other CPUs. 
But for SOPHGO RV SoC, GDB can debug all the CPUs on SoC. Is SOPHGO solely dependent on software implementation following?
>GDB inserting an ebreak into the program executed by CPU1
>When CPU1 encounters the ebreak, it traps to the kernel, halting the current program. 
>GDB then uses commands for debugging. //check the variables
>After debugging, it allows CPU1 to resume execution of the original program.

pdonahue-ventana, 2024-12-03:
There are multiple ways to set a breakpoint.  You can use the trigger module or you can replace an instruction in memory with an ebreak instruction (or a c.ebreak instruction).  This is very similar to how other architectures such as ARM do things (though the details obviously differ).

Using the trigger module would require the SBI [DBTR extension](https://github.com/riscv-non-isa/riscv-sbi-doc/blob/master/src/ext-debug-triggers.adoc).  I don't know whether SOPHGO supports that but I suspect that gdb is inserting ebreaks like you said.

closed 2024-12-16 by zhangdujiao