riscv/riscv-debug-spec #1090: Interrupt behaviour in debug mode

jackwdandrew, 2024-12-02:
My understanding of the intent of the specification is that:
* Interrupts are pended in the various interrupt controllers.
* Interrupts are masked in debug mode.
* Interrupts can be optionally masked in stepping.

The spec is not 100% clear:
* This [line](https://github.com/riscv/riscv-debug-spec/blame/bcd05b798f3584cc89aebb5c9e30012e65c711a1/Sdext.adoc#L36) only applies to abstract commands, should it be program buffer too?
* This [line](https://github.com/riscv/riscv-debug-spec/blob/bcd05b798f3584cc89aebb5c9e30012e65c711a1/debugger_implementation.adoc?plain=1#L76) is missing a cross reference to `stepie`
* There is no statement about whether interrupts continue to pend.

pdonahue-ventana, 2024-12-02:
> This [line](https://github.com/riscv/riscv-debug-spec/blame/bcd05b798f3584cc89aebb5c9e30012e65c711a1/Sdext.adoc#L36) only applies to abstract commands, should it be program buffer too?

The program buffer only executes as part of an abstract command:
 >> A debugger can write a small program to the Program Buffer, and then execute it exactly once with the Access Register Abstract Command

In fact, executing the program buffer is the only way to architecturally execute code in Debug Mode.  (Many implementations choose to execute code to communicate with the DM, to perform abstract commands besides the program buffer, etc.  However, this is should be done in a way that is indistinguishable from implementing hardware state machines and wires to/from DM.)

> This [line](https://github.com/riscv/riscv-debug-spec/blob/bcd05b798f3584cc89aebb5c9e30012e65c711a1/debugger_implementation.adoc?plain=1#L76) is missing a cross reference to stepie

You're right.  That looks like a latex->asciidoc conversion problem.  I created https://github.com/riscv/riscv-debug-spec/pull/1091.

> There is no statement about whether interrupts continue to pend.

I believe that the debug spec uses "disabled" in the same sense that the privileged spec uses enabled/disabled when describing interrupts.  A hart in debug mode couldn't possibly reach out to an interrupt controller to lower a pending interrupt.  That's normally done by a software handler and it involves more than just writing to some hart-local state.  (Even clearing the eip bit in a local IMSIC would leave the interrupt pending at the source.)

closed 2025-02-13 by pdonahue-ventana

tmoil, 2025-02-17:
> My understanding of the intent of the specification is that:
> 
> * Interrupts are pended in the various interrupt controllers.
> * Interrupts are masked in debug mode.
> * Interrupts can be optionally masked in stepping.
> 
> The spec is not 100% clear / correct:
> 
> * This [line](https://github.com/riscv/riscv-debug-spec/blame/bcd05b798f3584cc89aebb5c9e30012e65c711a1/Sdext.adoc#L36) only applies to abstract commands, should it be program buffer too?
> * This [line](https://github.com/riscv/riscv-debug-spec/blob/bcd05b798f3584cc89aebb5c9e30012e65c711a1/debugger_implementation.adoc?plain=1#L76) is missing a cross reference to `stepie`
> * There is no statement about whether interrupts continue to pend.

Hello, regarding the statement "Interrupts are masked in debug mode", does it mean that interrupts defined by other extensions should also be masked? For example, the RNMI interrupt defined in the Smrnmi extension.