riscv/riscv-debug-spec #1093: Does the base address of the 1st DM have to be at the bottom address space (0x000000)?

zhangdujiao, 2024-12-10:
From debug spec 3.1 "The bottom of the address space is used for the first (and usually only) DM"
We notice that OpenOCD can configure the DM address to be accessed via the **-dbgbase** command with any address in the cfg file, so it seems that there is no need to be at the very bottom of the address space. Is that ture?

en-sc, 2024-12-10:
`-dbgbase` is used as an quick-and-dirty solution.
The user does not need to know the exact address of the DM, the spec allows to discover this (based on the fact that the first DM is at the bottom and the address of the next one is in the `nextdm` register).

For additional info on OpenOCD's side of things see:
https://github.com/riscv-collab/riscv-openocd/issues/1147
https://github.com/riscv-collab/riscv-openocd/issues/278

pdonahue-ventana, 2024-12-10:
The answer to the question in the title is yes for the reason that you already quoted from the spec.  OpenOCD isn't the only debugger.

Even if you only want to use OpenOCD, you can only use -dbgbase if you already know something about what you're debugging.  One goal of a standardized debug architecture is that you can just connect a debugger to a chip and the debugger will be able to figure out the configuration without any hand-holding from the (possibly inexperienced) user.

closed 2024-12-13 by zhangdujiao