riscv/riscv-debug-spec #1102: "the address being access and which caused the trigger to fire"

tariqkurd-repo, 2025-01-24:
[image]

From this text where, for example, a `LW` loads bytes 0x4000, 0x4001, 0x4002, 0x4003
A watchpoint fires at address 0x4001
What is the reported VA value?

Is the address being accessed 0x4000? i.e. the calculated VA, and as a result the watchpoint fired.
Or does it mean the actual byte address needs to be reported i.e. 0x4001.

If it does mean the actual byte address, then it would help to clarify the wording.

pdonahue-ventana, 2025-03-14:
There are four addresses "being accessed" (in the first line of your example): 0x4000, 0x4001, 0x4002, 0x4003.
There is one address "which caused that trigger to fire" (in the second line of your example): 0x4001.
There is only one address which is "being accessed AND which caused that trigger to fire": 0x4001

On your NAPOT question: RISC-V doesn't actually specify the access order of non-atomic memory accesses.  If you LW from 0xFFE and both the 0xFFE page and the 0x1000 page get page faults, either one can trap first and tval can be either address (much to Allen Baum's chagrin).  It would probably be overstepping its bounds for the debug spec to further constrain the access order.  Therefore, it intentionally doesn't say which address is in tval.  It's whichever address the implementation decided is "being accessed" when the trigger first fired.

tariqkurd-repo, 2025-03-17:
ok - thanks @pdonahue-ventana - so:

* if only one byte causes a trigger to fire report the address of that byte
* if multiple causes cause a trigger to fire then it is legal to report any of them

thanks, that helps.

pdonahue-ventana, 2025-04-05:
This appears to have been answered satisfactorily.

closed 2025-04-05 by pdonahue-ventana