riscv/riscv-debug-spec #1107: Should the RNMI interrupt be masked in debug mode?

tmoil, 2025-02-17:
Hi，according to the description in the debug spec, "All interrupts (including NMI) are masked in debug". Does it mean that interrupts defined by other extensions should also be masked? For example, the RNMI interrupt defined in the Smrnmi extension.

pdonahue-ventana, 2025-02-17:
My initial reaction was that there is no "RNMI interrupt" because RNMI is a type of trap that is based on a normal NMI.  However, in reviewing the Smrnmi spec, it is very confusing.  It talks about "RNMI interrupt signals" and the "NMI trap handler" which is the opposite of my understanding.  But then it also talks about the "RNMI trap handler" which does align with my understanding.

@aswaterman: Is RNMI a distinct interrupt from NMI?  Can the Smrnmi spec be changed to use consistent terminology?

(In any case, "all interrupts" in the debug spec literally means all interrupts.  That includes all extensions.)

aswaterman, 2025-02-17:
RNMI is distinct from the implementation-defined concept of NMI defined in the privileged architecture spec (though I'm not sure it makes sense for both to exist in the same implementation).

tmoil, 2025-02-18:
> My initial reaction was that there is no "RNMI interrupt" because RNMI is a type of trap that is based on a normal NMI. However, in reviewing the Smrnmi spec, it is very confusing. It talks about [...]

Okay，i got it. Thanks

pdonahue-ventana, 2025-02-19:
I guess that RNMI can be considered an entirely different interrupt and not just a different hart behavior upon assertion of the same old NMI interrupt input.  If that's the case, the references to "NMI" in Smrnmi (other than the first paragraph that discusses legacy NMI) seem incorrect.  For example: "The mnscratch CSR holds an MXLEN-bit read-write register which enables the NMI trap handler to save and restore the context that was interrupted."  The mnscratch CSR has to do with the RNMI trap handler, not the legacy/implementation-defined NMI handler.

aswaterman, 2025-02-19:
I agree.  I'll fix them.

closed 2025-02-20 by aswaterman