riscv/riscv-debug-spec #1108: Dose mstatus.mprv affect the trigger tdata1 fire mode？

AlexGJL, 2025-02-19:
Trigger register tdata1(e.g. mcontrol6.s) make trigger must fire in S/HS-mode. but mstatus.mprv could affect the “privilege mode”，1) dose mstatus.mprv affect the mode that trigger fires in? 2)do dcsr.mprven and mnstatus.nmie affect the mode that trigger fires in?

AlexGJL, 2025-02-20:
Thank you for your replay.

  form nmie, There is such a description in The RISC-V Instruction Set Manual: Volume II： "When NMIE=0, the hart behaves as though mstatus.MPRV were clear, regardless of the current setting of mstatus.MPRV." ，which means it could affect mstatus.MPRV work。as the mode filters in tdata1 refer to the actual mode, nmie also not affect  triggers fire.

pdonahue-ventana, 2025-02-20:
Now your question makes sense to me.  But the answer to question 1 makes the dcsr.mprven and mnstatus.NMIE questions moot.

pdonahue-ventana, 2025-04-05:
This question seems to be answered.  You might also be interested in the Sdtrigepm proposal:  https://docs.google.com/document/d/1itWuxRKILehYoyRMWLgWF71dyEwZcikvTRj1DPCn6x0/edit?tab=t.0#heading=h.qg2p9shauf6

closed 2025-04-05 by pdonahue-ventana