riscv/riscv-debug-spec #1111: minhibit/mcounteren

jiahzhang, 2025-03-04:
Should minhibit/mcounteren affect how icount works?

pdonahue-ventana, 2025-03-05:
No.  Those registers affect HPM and counter CSRs which are completely separate from Sdtrig.

You might be thinking that icount and minstret are related.  However, icount must count both retires and traps based on various filters (if present): mode in icount, ASID in tdata3, VMID in tdata3, scontext in tdata3, hcontext in tdata3.  On the other hand, minstret must count only retires (not traps) based on a separate set of filters: mcountinhibit and minstretcfg (if Smcntrpmf is implemented).  Obviously, the "an instruction has retired" logic may be shared but different things happen after that.

jiahzhang, 2025-03-05:
Understood, thanks

closed 2025-03-05 by jiahzhang