riscv/riscv-debug-spec #1112: Questions about the implementation of the mcontrol6.match field

tmoil, 2025-03-11:
Hi，According to the description in the debug spec, the values of mcontrol6.match are 0, 1, 2, 3, 4, 5, 8, 9, 12, 13 and others that are reserved for future use. When implementing  in the project, which values (matching modes) must be implemented? And which values (matching modes) are optionally implemented? In my understanding, according to the project requirements, it is sufficient to implement one or more of 0, 1, 2, 3, 4, 5, 8, 9, 12, 13. The debug spec does not specify which value must be implemented.

pdonahue-ventana, 2025-03-11:
In Sdtrig, no match values are mandatory and all values are optional:
> Almost all trigger functionality is optional. All tdata registers follow write-any-read-legal semantics. If
a debugger writes an unsupported configuration, the register will read back a value that is supported
(which may simply be a disabled trigger). This means that a debugger must always read back values it
writes to tdata registers, unless it already knows what is supported. 

This provides maximum flexibility for implementations to provide the right set of features for their needs.  If someone has some strange reason to only implement match=13 then that's fine.  Using the normal WARL mechanism, a debugger will discover this and either use the features that are provided or decide that it can't use this trigger.

This flexibility in Sdtrig (and elsewhere in RISC-V) doesn't mean that requirements can't be imposed by some other spec like a profile or the now-abandoned [platform spec](https://github.com/riscvarchive/riscv-platform-specs/blob/main/riscv-platform-spec.adoc?plain=1#L270) (which was not ratified but which would have mandated at least match=0 support).

tmoil, 2025-03-12:
Thanks，i got it

closed 2025-03-28 by pdonahue-ventana