riscv/riscv-debug-spec #1114: About uncertain match for unit-stride Fault-Only-First Loads

qin996, 2025-03-24:
If we use uncertain watchpoint match for vleff Instructions,  there are still some doubts.

for vleff instructions, there is an example for 'strlen' in spec:
`strlen:
 mv a3, a0 # Save start
loop:
 vsetvli a1, x0, e8, m8, ta, ma # Vector of bytes of maximum length
 vle8ff.v v8, (a3) # Load bytes
 csrr a1, vl # Get bytes read
 vmseq.vi v0, v8, 0 # Set v0[i] where v8[i] = 0
 vfirst.m a2, v0 # Find first set bit
 add a3, a3, a1 # Bump pointer
 bltz a2, loop # Not found?
 add a0, a0, a1 # Sum start + bump
 add a3, a3, a2 # Add index
 sub a0, a3, a0 # Subtract start address+bump`

`These instructions execute as a regular load except that they will only take a trap caused by a synchronous exception on element 0. If element 0 raises an exception, vl is not modied, and the trap is taken. If an element > 0 raises an exception, the corresponding trap is not taken, and the vector length vl is reduced to the index of the element that would have raised an exception.`

and for mcontrol6 address triggers, debug spec describes that
`0 (address): There is at least one compare value and it contains the lowest virtual address of the access.`

So, for address triggers, can we only determine if there is a watchpoint uncertain match on the first element? because If the watchpoint address match is judged on elements other than element0, the address will change during the instruction resend process and may not trigger a breakpoint fault again.

There is another more corner scenario, where if the vleff instruction accesses across pages, it is often split into two parts in micro-architecture implementation. Can we only compare the addresses of the element0 in first part, because the lowest virtual address accessed must occur in the first part of the cross page.

pdonahue-ventana, 2025-03-24:
The "lowest virtual address of the access" is referring to the access, not the instruction.  Vector load/store instructions do many accesses.  Even a misaligned `lw` instruction might do multiple accesses (depending on the implementation and what NAPOT regions it crosses, such as crossing pages in your example).

The V extension's description of fault-only-first instructions specifically talks about mcontrol6:
> When the fault-only-first instruction would trigger a debug data-watchpoint trap on an element after the first, implementations should not reduce vl but instead should trigger the debug trap as otherwise the event might be lost.

So mcontrol6 behaves like with other vector instructions and doesn't do the special vl reduction behavior of fault-only-first.

qin996, 2025-03-25:
> When the fault-only-first instruction would trigger a debug data-watchpoint trap on an element after the first, implementations should not reduce vl but instead should trigger the debug trap as otherwise the event might be lost.

spec only describes data-watchpoint for fault-only-first, so for address triggers, is it the same? 
Should the  implementations not reduce vl but instead trigger the debug trap?

pdonahue-ventana, 2025-03-25:
> 1. spec only describes data-watchpoint for fault-only-first, so for address triggers, is it the same? Should the implementations not reduce vl but instead trigger the debug trap?

A "data-watchpoint" is using a relatively common industry term for watching data (not instruction) accesses to a particular memory location (e.g. reading/writing a variable).  It normally would involve what the debug spec calls load/store address triggers but I suppose that it could also involve load/store data triggers.

> 2. misaligned scalar Instructions(eg. lw), back to back access memory in micro-architecture implementation, however it is actually seen is only one memory access for software. So can we consider as only one memory access for address triggers in this case?

If nothing can possibly come between the back-to-back accesses then they are atomic and essentially a single access.  If something can come between them then they're not atomic and definitely not a single access.  For instance, they might go down the pipe back-to-back but the second one can miss in the cache.  Even if the hart blocks and cannot be interrupted until the data is returned, the two accesses were not necessarily back-to-back in global memory order and can therefore be observed as two distinct accesses.  This has more to do with RVWMO/RVTSO than debug.  For debug purposes, anything that is guaranteed by the memory consistency model to be atomic will involve a single access and anything that is not guaranteed to be atomic has an implementation-defined number of accesses (one or more).

> 3. for vle and vleff instructions, a memory access is also performed once in the specific implementation. Take for example doing a 16B unit-stride load from cacheable main memory, An implementation will simply do a full 16B access. so if the "lowest virtual address of the access" is referring to the access, can we only compare the address of element0 with tdata2 for address trigger?

I think that your question is whether you're **required** to compare **only** tdata2 vs. the address of element 0.  No.

As select=0 says, "There is **at least one** compare value and it contains the lowest virtual address of the access. It is **recommended that there are additional compare values** for the other accessed virtual addresses."  A minimal implementation will do what you said.  A recommended implementation will fire the trigger if any of the 16 byte addresses matches tdata2, but that's not required.  When the uncertain bit is considered, an implementation could also cause the trigger to fire if none of the 16 byte addresses matches tdata2 (but hopefully it matches some nearby address).

In summary:
- You are not required to only compare the address of element 0 (i.e. you are not required to compare no other addresses).
- You are allowed to only compare the address of element 0.
- You are required to compare at least the address of element 0.

closed 2025-03-26 by qin996