riscv/riscv-debug-spec #1115: question regarding dcsr.stepie

Wen-Tian-Pineapple, 2025-03-26:
I'm a bit confused by the description of this part of DCSR, what does interrupt can be emulated by debugger mean when interrupt is disabled in debug mode? Is there any examples? Thanks.

pdonahue-ventana, 2025-03-26:
dcsr.stepie controls whether interrupts can be taken while single-stepping.  If that bit is hardwired to 0 then interrupts can never be taken while single-stepping.  If interrupts are not taken for a long period of time then some systems might have problems.

This sentence says that the debugger can notice that interrupts are pending (probably by reading mip) and it can emulate the interrupt behavior, much like privileged software can emulate things on behalf of less privileged modes.

pdonahue-ventana, 2025-04-05:
I believe that this has been answered.

closed 2025-04-05 by pdonahue-ventana