riscv/riscv-debug-spec #1118: icount trigger matching when MIE is modified

jiahzhang, 2025-04-09:
The spec says
```
This trigger matches when:
1. An instruction retires after having been fetched in a privilege mode where the trigger is enabled. This explicitly includes all RET instructions from various modes.
2. A trap is taken from a privilege mode where the trigger is enabled. This explicitly includes traps taken due to interrupts.
```
Should it also match if the instruction retired disables MIE?

pdonahue-ventana, 2025-04-10:
I assume that this question is in relation to action=0 triggers as described in the Native Triggers section.  Note that icount is the highest priority synchronous exception on the next instruction.  By the time that we're executing the next instruction, MIE is already updated and icount should definitely not fire.  But that's not exactly your question.

As for whether it should match (and set the pending bit which might then cause it to fire at some future time), the RISC-V execution model is not well-defined so it's unclear whether:
1. MIE is updated before retire, in which case the match must be suppressed
2. MIE is updated after retire, in which case the match must not be suppressed
3. MIE is updated concurrent with retire, in which case the debug spec should explicitly say whether the match is suppressed, the match is not suppressed, or the match behavior is UNSPECIFIED

The ISA itself provides no guidance between the three and the debug spec doesn't say which of the three might apply (or which of the three sub-choices applies if option 3 applies).  The debug spec is ratified so we cannot introduce any change that might not be compatible with an existing implementation.  Because existing compliant implementations might either match or not match, it seems to me that the unspecified behavior should be officially UNSPECIFIED.

jiahzhang, 2025-04-10:
Thanks for the response

closed 2025-04-10 by jiahzhang