riscv/riscv-debug-spec #1119: Why cache operations encoded as HINTs do not match debug triggers?

qin996, 2025-04-21:
in debug spec, it describes that
> Cache operations encoded as HINTs do not match debug triggers.

Why cache operations encoded as HINTs do not match debug triggers? I don't know the reason about this, who can help me? Thanks

pdonahue-ventana, 2025-04-21:
The unprivileged spec says:
> Like the NOP instruction, HINTs do not change any architecturally visible state, except for advancing the pc and any applicable performance counters. Implementations are always allowed to ignore the encoded hints.

Trapping or halting (or other things like turning trace on or off) would be changing architecturally visible state.  It's similar to the reason that the CMO chapter says that the Zicbop HINTs aren't allowed to generate page faults or access faults.

qin996, 2025-04-22:
Does the cache operations encoded as HINTs only contain Zicbop instructions?
The unprivileged spec in Zihintntl chapter says:
> The NTL instructions can affect cache-management operations other than those in the Zicbom extension. For example, NTL.PALL followed by CBO.ZERO might indicate that the line should be allocated in L3 and zeroed, but not allocated in L1 or L2.

So in this case, if NTL.PALL followed by CBO.ZERO, should cbo.zero match debug triggers?

pdonahue-ventana, 2025-04-22:
I believe that the only CMOs that are HINTs are the ones in Zicbop.

Your example consists of two instructions: NTL.PALL followed by CBO.ZERO.  The NTL.PALL instruction is a HINT which means that it (1) does not update any architectural state and (2) does not match debug triggers (which it wouldn't do anyway because it doesn't access memory).  The CBO.ZERO is a normal instruction and the debug spec says that it "matches as if it is a store."

If your implementation decodes NTL.PALL+CBO.ZERO as a single 64b instruction, it must behave as if those were two separate instructions for purposes of traps and debug triggers.  Just like a page fault on CBO.ZERO must have epc point to the CBO.ZERO instead of the NTL.PALL, a debug trigger must behave normally on the CBO.ZERO and epc/dpc must point to the CBO.ZERO.

closed 2025-04-23 by qin996