riscv/riscv-debug-spec #1121: Some issues regarding the transfer and postexec fields of abstract commands

tmoil, 2025-04-28:
Hi， 
    the following description can be found in the debug specification：

[image]

And，when I was debugging in an project using OpenOCD, the following situation occurred in the openocd.log: The configuration command is set to 0x241000:

[image]

I have the following questions:

Q1:  According to the description in the debug spec, when the command is configured as 0x241000, it should be judged as a failure. However, judging from the behavior of OpenOCD, it expects the hardware to recognize this kind of command with transfer being 0 and postexec being 1 as a legal abstract command and execute the program in the progbuf normally. I also agree with this expectation of OpenOCD. Is this understanding correct?

Q2: Furthermore, if you also agree with the view in Q1, then for a command with both transfer and postexec set to 0 configured by the debugger (assuming the debugger would do so), should the hardware detect and report a failure (cmderr)? If it reports a failure, should the cmderr code be 2 (not supported)?

thanks

pdonahue-ventana, 2025-04-28:
> Q1: According to the description in the debug spec, when the command is configured as 0x241000, it should be judged as a failure because its transfer field is 0. However, judging from the behavior of OpenOCD, it expects the hardware to recognize this kind of command with transfer being 0 and postexec being 1 as a legal abstract command and execute the program in the progbuf normally. I also agree with this expectation of OpenOCD. Is this understanding correct?

I don't understand your reasoning.  The spec says:
1. "If write is clear and transfer is set, then ..."  That is false because transfer=0 so we don't do the operation specified in the "then" clause.
2. "If write is set and transfer is set, then ..."  That is false because transfer=0 so we don't do the operation specified in the "then" clause.
3. "If aarpostincrement and transfer are set, ..."  That is false because transfer=0 so we don't do the operation specified in the "then" clause.  (If you look at the definition of aarpostincrement, it's actually undefined whether we do that increment operation but I don't think that it's possible for the increment to fail anyway.)
4. "Execute the Program Buffer, if postexec is set."  That bit is set so we perform this step.

This command is very commonly used when you don't want to transfer anything and you just want to execute the program buffer.  The spec suggests that a value of 0 on transfer "can be used to just execute the Program Buffer."

> Q2: Furthermore, if you also agree with the view in Q1, then for a command with both transfer and postexec set to 0 configured by the debugger (assuming the debugger would do so), should the [...]

None of the "if" conditions in the 4 steps would be true so we wouldn't perform any of the 4 behaviors.  If we don't perform an operation then the operation cannot fail.  Such a command would just be a nop and would never fail.

tmoil, 2025-04-29:
It's possible that I misinterpreted the "fail" described in the following  of the spec as the situation where the "if" conditions in the above four steps are not met. 

[image]

Such an understanding is indeed problematic, and based on this, my reasoning logic is also wrong. 
Thank you very much for your explanation. I've got it now.