riscv/riscv-debug-spec #1124: How to use relaxedpriv in abstractcs?

shengqilong, 2025-05-11:
```
This optional bit controls whether program buffer and abstract memory accesses are performed with the exact and full set 
of permission checks that apply based on the current architectural state of the hart performing the access, or with a relaxed 
set of permission checks (e.g. PMP restrictions are ignored). The details of the latter are implementation-specific.
```
How to understand the above passage, and how should relaxedpriv be used specifically?

pdonahue-ventana, 2025-05-11:
The bit is provided for implementations that (for whatever reason) cannot apply PMP or other checks on PB and abstract memory accesses.  It's optional and I would expect that almost everyone only supports relaxedpriv=0.  Note that Sdsec will require that relaxedpriv=1 must not be supported.

shengqilong, 2025-05-13:
Can you give an example for which specific situations for “other checks on PB”

shengqilong, 2025-05-13:
Is this referring to the access check for PB itself, or the permission check for instruction execution within PB?

closed 2025-09-16 by pdonahue-ventana