riscv/riscv-debug-spec #1128: While an abstract command is executing , should write Abstract Command or Abstract Comtrol and Status be ignored?

tmoil, 2025-08-07:
Hi,
In the latest debug spec，when an abstract command is being executed, write operations to registers such as abstractauto, progbuf, and abstract data will be ignored. 

[image]

[image]

When abstractcs.busy is valid，can the write operations to the abstract command and abstractcs register be ignored ?

[image]

thanks

pdonahue-ventana, 2025-08-08:
As you say, attempts to write progbuf0 while busy is set does not change its value.  The command register is WARZ so there's no need to define whether the value changes because the value is always 0.

tmoil, 2025-08-08:
What about the abstractcs register, especially the cmderr field, which is W1C (Write 1 to Clear) ?  If software writes to abstractcs while an abstract command is being executed—particularly writing all 1s to cmderr—would this write operation cause unpredictable effects on the process where the ongoing abstract command is updating cmderr?

[image]

[image]

thanks

pdonahue-ventana, 2025-08-08:
Can you explain a scenario where this would matter?  I might be missing something but I'm not sure how you can have busy=1 and cmderr!=0.  This combination means that we're still busy executing a command and an error has already occurred.

If cmderr!=0 then no abstract command can start so we can't become busy.  That means that to have busy=1 and cmderr!=0, the cmderr would have to be set after the current command began.  But I believe that cmderr cannot be written by hardware unless busy=0 (either because the unsupported command never began or because a command has just completed).

If cmderr=0 then it doesn't matter whether those bits get cleared because they're already clear.

tmoil, 2025-08-11:
> That means that to have busy=1 and cmderr!=0, the cmderr would have to be set after the current command began. But I believe that cmderr cannot be written by hardware unless busy=0 (either because the unsupported command never began or because a command has just completed).
When the core finishes executing an abstract command, it notifies the DM (Debug Module) of the execution result. If an exception occurs during the execution of the abstract command, the DM, upon learning this, will clear the busy and set cmderr to 3. At the same time, if an write from debugger happens to abstractcs.cmderr (writing 1s) and this write operation is not ignored by the hardware, should cmderr be 3 or 0? Should we impose a constraint that software should not initiate write operations to abstractcs when busy is 1?