riscv/riscv-debug-spec #1130: The impact of H extension implementation on registers such as tdata3.mhselect, hcontext, and mcontrol6.vs

tmoil, 2025-08-12:
In the debug spec, there is the following description regarding the implementation of some fields of the register:
1. tdata3.mhselect：

[image]

2. hcontext：

[image]

3. mcontrol6.vs：

[image]

Regarding the aforementioned "not supported" and "not implemented", I have the following two questions:
(1) Do "not supported" and "not implemented" refer to the fact that the relevant logic for the H extension is not implemented in our specific hardware implementation, or that the hardware implements the H extension but the software disables it by clearing misa.H?
(2) If the hardware implements the H extension, will these fields change in bit width or value as misa.H is set or cleared during software execution? And is this change implicit in the hardware?

tmoil, 2025-08-13:
Thank you for your explanation. It's clear to me now.

pdonahue-ventana, 2025-08-13:
I'm happy to help.

closed 2025-08-13 by pdonahue-ventana